MarlinFirmware/Marlin #15316: [BUG] MKS Robin TFT 1.0 Touch not working.

1uknowleast, 2019-09-20:
### Description

MKS Robin TFT 1.0 Touch not working on a MKS Robin Mini board.

I have a Kingroon 3DP180 printer that has a MKS Robin Mini 2 controller and MKS Robin TFT 1.0 Touch Screen.

If I compile the branch of Marlin 2.0 from https://github.com/berghauz/Marlin-bugfix-2.0.x-FLSUN-QQ-S the touch screen works with out issue. However that version has stuttering when Linear advance is enabled and it from May 2019, so quite old in Marlin Years. Going back to the original Smoothieware firmware also works fine.

I am able to compile the Marlin 2.0 bugfix from 2019-09-18 and flash the board. Controlling the printer via Octoprint and USB shows now issues. I see the TFT Screen power on, and show information, however the touch doesn't respond in any way.

I've tried various XPT2046 CALIBRATION settings from various source, none seem to make the difference.

Im hoping its just a simple software configuration issue. Your help is much appreciated.

**Expected behavior:** Touch Screen to respond to touch

**Actual behavior:** Touch Screen displays information, but does not respond to touch

#### Additional Information

[image]
[image]
[configs-MKS_ROBIN_TFT24_1.0_not_working.zip](https://github.com/MarlinFirmware/Marlin/files/3636703/configs-MKS_ROBIN_TFT24_1.0_not_working.zip)

1uknowleast, 2019-09-21:
I should add that enabling a filament runout sensor results in firmware that doesn't boot on the board as well. After flashing just a black screen, and no response on the USB port.

boelle, 2019-10-12:
@1uknowleast still having issues?

boelle, 2019-10-24:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-24 by boelle