MarlinFirmware/Marlin #15320: [BUG] (short description)

thetechy2, 2019-09-20:
### Description
 Im setting up an skr 1.3 with tmc2208s with marlin 2.0 on a corexy im building. I have an x max endstop; so i set  X_HOME_DIR to 1 to have it home in the max direction, but when i try to home the x axis from the lcd menu it moves in the min direction and only a few mm

### Steps to Reproduce

1. have x max endstop
2. set home direction positive (and enable individual axis homing)
3. upload
4. home x axis

**Expected behavior:** x to home on the max endstop

**Actual behavior:** x carriage moves a few mms in the min direction

#### Additional Information

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/3637993/configs.zip)

reloxx13, 2019-09-21:
does x move in the correct direction when you move it by lcd?   
its normal to move away from the endstop at beginning and then it starts to move to the endstop.    
if it does not start moving to the endstop after going away, it thinks the endstop is triggered already.   
sounds for a hardware issue to me.

thetechy2, 2019-09-22:
x moves correctly yes
when i set the endstop to min direction in the firmware it moved accordingly and i had to shut it off to avoid crashing
then when i change it back to max direction it only moves a few mm at a time and in the min direction

thetechy2, 2019-09-22:
i got it

closed 2019-09-22 by thetechy2