MarlinFirmware/Marlin #15322: [BUG] Compiling error

npmomchev, 2019-09-21:
### Description

if you allow "#define G29_RETRY_AND_RECOVER" and "#define HOST_PROMPT_SUPPORT" at the same time, compiling failed with error: 
Marlin\src\Marlin.cpp: In function 'void event_probe_recover()':
Marlin\src\Marlin.cpp:417:22: error: 'PROMPT_INFO' was not declared in this scope
       host_prompt_do(PROMPT_INFO, PSTR("G29 Retrying"));
### Steps to Reproduce

1. [Enable: #define G29_RETRY_AND_RECOVER]
2. [Enable: #define HOST_PROMPT_SUPPORT]
3. [Compiling...]

[image]
[image]

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3638249/Marlin.zip)

sjasonsmith, 2019-09-21:
PROMPT_INFO was removed a couple weeks ago. This line of code was probably supposed to be deleted.

boelle, 2019-10-12:
@npmomchev tried with latest copy of 2.0.x ?

boelle, 2019-10-21:
will close this one as it seems @npmomchev dont have an issue anymore

we can always reopen if its not the vase

closed 2019-10-21 by boelle