MarlinFirmware/Marlin #15324: [BUG] Compiling error

PhoenixbgOrg, 2019-09-21:
Im using SKR pro 1.1 with TMC2209 on last Marlin 2.0

[image]

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/3638500/Config.zip)

reloxx13, 2019-09-21:
just to confirm, you hitted "clean" before compiling?

[image]

PhoenixbgOrg, 2019-09-22:
Yes i try all this

forthereg, 2019-09-24:
Im also having issue compiling the new marlin for skr PRO 

i even tried to use the config/adv config/ platformio/ini file from my last marlin and got the same errors.

Compiling .pio\build\BIGTREE_SKR_PRO\FrameworkArduino\Tone.cpp.o
In file included from C:\Users\Helix\.platformio\packages\framework-arduinoststm32\variants\BIGTREE_GENERIC_STM32F407_5X\variant.cpp:31:0:
C:\Users\Helix\.platformio\packages\framework-arduinoststm32\cores\arduino/pins_arduino.h:24:15: error: expected constructor, destructor, or type conversion before '(' token
 _Static_assert(LastPort <= 0x0F, "PortName must be less than 16");
               ^
C:\Users\Helix\.platformio\packages\framework-arduinoststm32\cores\arduino/pins_arduino.h:53:15: error: expected constructor, destructor, or type conversion before '(' token
 _Static_assert(NUM_ANALOG_INPUTS <= MAX_ANALOG_INPUTS,
               ^
C:\Users\Helix\.platformio\packages\framework-arduinoststm32\cores\arduino/pins_arduino.h:55:15: error: expected constructor, destructor, or type conversion before '(' token
 _Static_assert(NUM_ANALOG_FIRST >= NUM_ANALOG_INPUTS,
               ^
C:\Users\Helix\.platformio\packages\framework-arduinoststm32\cores\arduino/pins_arduino.h:63:14: error: 'uint8_t' does not name a type
 static const uint8_t A0 = PIN_A0;
              ^~~~~~~
C:\Users\Helix\.platformio\packages\framework-arduinoststm32\cores\arduino/pins_arduino.h:67:14: error: 'uint8_t' does not name a type
 static const uint8_t A1 = PIN_A1;
              ^~~~~~~
compilation terminated due to -fmax-errors=5.
Compiling .pio\build\BIGTREE_SKR_PRO\FrameworkArduino\USBSerial.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\FrameworkArduino\WInterrupts.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\FrameworkArduino\WMath.cpp.o
*** [.pio\build\BIGTREE_SKR_PRO\FrameworkArduinoVariant\variant.cpp.o] Error 1
Compiling .pio\build\BIGTREE_SKR_PRO\FrameworkArduino\WSerial.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\FrameworkArduino\WString.cpp.o
In file included from C:\Users\Helix\.platformio\packages\framework-arduinoststm32\cores\arduino\HardwareSerial.h:30:0,
                 from C:\Users\Helix\.platformio\packages\framework-arduinoststm32\cores\arduino\WSerial.h:5,
                 from C:\Users\Helix\.platformio\packages\framework-arduinoststm32\cores\arduino\WSerial.cpp:1:
C:\Users\Helix\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/uart.h:64:3: error: 'PinName' does not name a type; did you mean 'rename'?
   PinName pin_tx;
   ^~~~~~~
   rename
C:\Users\Helix\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/uart.h:65:3: error: 'PinName' does not name a type; did you mean 'rename'?
   PinName pin_rx;
   ^~~~~~~
   rename
In file included from C:\Users\Helix\.platformio\packages\framework-arduinoststm32\cores\arduino\WSerial.h:5:0,
                 from C:\Users\Helix\.platformio\packages\framework-arduinoststm32\cores\arduino\WSerial.cpp:1:
C:\Users\Helix\.platformio\packages\framework-arduinoststm32\cores\arduino\HardwareSerial.h:106:28: error: expected ')' before '_rx'
     HardwareSerial(PinName _rx, PinName _tx);
                            ^~~
C:\Users\Helix\.platformio\packages\framework-arduinoststm32\cores\arduino\HardwareSerial.h:126:16: error: 'PinName' has not been declared
     void setRx(PinName _rx);
                ^~~~~~~
C:\Users\Helix\.platformio\packages\framework-arduinoststm32\cores\arduino\HardwareSerial.h:127:16: error: 'PinName' has not been declared
     void setTx(PinName _tx);
                ^~~~~~~
*** [.pio\build\BIGTREE_SKR_PRO\FrameworkArduino\WSerial.cpp.o] Error 1
In file included from C:\Users\Helix\.platformio\packages\framework-arduinoststm32\cores\arduino\Arduino.h:48:0,
                 from C:\Users\Helix\.platformio\packages\framework-arduinoststm32\cores\arduino\Tone.cpp:22:
C:\Users\Helix\.platformio\packages\framework-arduinoststm32\cores\arduino\Tone.cpp: In function 'void tone(uint8_t, unsigned int, long unsigned int)':
C:\Users\Helix\.platformio\packages\framework-arduinoststm32\cores\arduino\pins_arduino.h:209:73: error: 'digitalPin' was not declared in this scope
 #define digitalPinToPinName(p)      (((uint32_t)p < NUM_DIGITAL_PINS) ? digitalPin[p] : NC)
                                                                         ^
C:\Users\Helix\.platformio\packages\framework-arduinoststm32\cores\arduino\Tone.cpp:31:15: note: in expansion of macro 'digitalPinToPinName'
   PinName p = digitalPinToPinName(_pin);
               ^~~~~~~~~~~~~~~~~~~
C:\Users\Helix\.platformio\packages\framework-arduinoststm32\cores\arduino\pins_arduino.h:209:73: note: suggested alternative: 'digitalRead'
 #define digitalPinToPinName(p)      (((uint32_t)p < NUM_DIGITAL_PINS) ? digitalPin[p] : NC)
                                                                         ^
C:\Users\Helix\.platformio\packages\framework-arduinoststm32\cores\arduino\Tone.cpp:31:15: note: in expansion of macro 'digitalPinToPinName'
   PinName p = digitalPinToPinName(_pin);
               ^~~~~~~~~~~~~~~~~~~
C:\Users\Helix\.platformio\packages\framework-arduinoststm32\cores\arduino\Tone.cpp: In function 'void noTone(uint8_t)':
C:\Users\Helix\.platformio\packages\framework-arduinoststm32\cores\arduino\pins_arduino.h:209:73: error: 'digitalPin' was not declared in this scope
 #define digitalPinToPinName(p)      (((uint32_t)p < NUM_DIGITAL_PINS) ? digitalPin[p] : NC)
                                                                         ^
[... cut ...]
C:\Users\Helix\.platformio\packages\framework-arduinoststm32\cores\arduino\pins_arduino.h:209:73: note: suggested alternative: 'digitalRead'
 #define digitalPinToPinName(p)      (((uint32_t)p < NUM_DIGITAL_PINS) ? digitalPin[p] : NC)
                                                                         ^
C:\Users\Helix\.platformio\packages\framework-arduinoststm32\cores\arduino\WInterrupts.cpp:64:15: note: in expansion of macro 'digitalPinToPinName'
   PinName p = digitalPinToPinName(pin);
               ^~~~~~~~~~~~~~~~~~~
*** [.pio\build\BIGTREE_SKR_PRO\FrameworkArduino\WInterrupts.cpp.o] Error 1

forthereg, 2019-09-24:
alot of 
C:\Users\Helix\.platformio\packages\framework-arduinoststm32\cores\arduino\pins_arduino.h:209:73: error: 'digitalPin' was not declared in this scope

xeddog, 2019-09-24:
I am having the same problem as the OP.  Same error messages, but I have 2208's instead of 2209's.  I've been trying to figure this one out for about 3 days.

xeddog, 2019-09-25:
Update:   I ran a pull this morning (Sept 25) and received over 300 updates.  Then did a "clean", deleted the .pio and .vscode directories, and ran a build.  Still failed with the same errors.

sjasonsmith, 2019-09-26:
As a temporary solution, you can continue building against the BigTreeTech version of TMCStepper. You will need to edit `trinamic.h` in  Marlin and comment out the version check.

xeddog, 2019-09-26:
I remembered that last night after my last post.  I updated my platformio.ini to do that, but I still got the same errors.  I intend to go over that again today to make sure I did that correctly, but I think I did.

xeddog, 2019-09-27:
OK, I did have an error in my platformio.ini file.  It took me a few tries to nail it down, but I got a good compile using the BTT stepper library.

I know you guys are quite busy now, but any idea when the corresponding Marlin upgrades might happen?

xeddog, 2019-09-29:
Ran another pull yesterday (09-28) and the platformio.ini updated the tmcstepper library to include >=0.5.1.  Build still fails as before and I had to go back to the BTT version.

thinkyhead, 2019-10-14:
@xeddog — If you're still experiencing the issue, post your configs and we'll make sure to get it sorted asap.

xeddog, 2019-10-15:
I have attached two files.  A zip filed which contains the Marlin configuration information including configuration.h, configuration.adv.h, the bigtreepro pins file, and the platformio.ini.  Also included is a separate file that shows the compile errors I get.  

Thanks,

Wayne

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3730447/Configuration.zip)
[TMC2208 Build errors.txt](https://github.com/MarlinFirmware/Marlin/files/3730449/TMC2208.Build.errors.txt)

boelle, 2019-10-24:
@PhoenixbgOrg did hitting clean sort the issue?

xeddog, 2019-10-24:
> 
> 
> @PhoenixbgOrg did hitting clean sort the issue?

No.  I run the clean, and sometimes even delete the .pio and .vscode folders too. and still get the build errors.

thinkyhead, 2019-10-24:
I've been able to get past most of your build errors but `TMCStepper` is still complaining about bad function calls.

Updated configs…

[configs.tar.gz](https://github.com/MarlinFirmware/Marlin/files/3769360/configs.tar.gz)

xeddog, 2019-10-26:
> 
> 
> I've been able to get past most of your build errors but `TMCStepper` is still complaining about bad function calls.
> 
> Updated configs…
> 
> [configs.tar.gz](https://github.com/MarlinFirmware/Marlin/files/3769360/configs.tar.gz)

I went through the configs you included and did not see any changes to mine.  Any hints??

boelle, 2019-10-26:
try and use a tool like winmerge to compare

xeddog, 2019-10-26:
OK, I downloaded winmerge and compared files.  The only changes were a result of a recent pull and had nothing to do with TMC drivers.  Configuration.h was one line added dealing with bootscreen, and configuration.adv.h was a few lines dealing with Z-auto-align.

boelle, 2019-10-26:
yep, i use it if i have not updated in a while and i need to have my settings moved from old file to new

boelle, 2019-11-03:
will close this one as it seems @PhoenixbgOrg have no issues

we can reopen if several can confirm an issue with marlin itself

closed 2019-11-03 by boelle