MarlinFirmware/Marlin #15326: [BUG] firmware.bin does not change to FIRMWARE.CUR

kurtisane, 2019-09-21:
Hi,
I'm having a issue with the SKR v1.3. I was able to just change the firmware with putting the firmware.bin on the SD Card and inserting it, then starting it up. When I put the SD Card back in the PC I see the FIRMWARE.CUR. 

But since today the firmware.bin does not change to FIRMWARE.CUR.

I'm not able to find out why. 

Regards

p3p, 2019-09-21:
The only reasons it wouldn't flash would be hardware or a corrupt firmware.bin because you didn't use safely remove hardware when you removed the sd card from the computer, the bootloader that handles flashing never changes and is not part of Marlin.

Make sure the card and socket are clean, the sdcard filesystem is not corrupt, also it sometimes helps if you hold the reset button for a few seconds before releasing when trying to flash.

reloxx13, 2019-09-21:
what p3p said, format the sd card and try again or use another higher quality sd card.

set a version in un ur machine name (in the config file), that way you caan see if it flashed successfull.

igorepi, 2019-09-22:
Another possible reason is damaged (or missing) LPC1768 DFU bootloader. You may need reflash DFU bootloader via serial bootloader/jtag/etc.

p3p, 2019-09-22:
Although possible, I have never seen a case of this bootloader being damaged by flashing the application area, or in general use.

igorepi, 2019-09-22:
Well, I personally managed to kill DFU bootloader on my SKR 1.3 while debugging Marlin with jlink.

kurtisane, 2019-09-22:
I guess the bootloader is dead.

I'm not able to find any guide how to reflash it.

p3p, 2019-09-22:
@igorepi was using jlink to flash directly, if you have only used the sdcard method you haven't damaged the bootloader.

SSPW, 2019-09-26:
I have this problem with SKR E3 DIP.
I have build firmware multiple ways.
But erase of firmware.bin never happens.
Creation of FIRMWARE.CUR is not consistent.
But flashing does work.
I have to erase the firmware.bin after flash, so it doesn't reflash every boot.

han9988, 2019-10-08:
I have a bunch of these 128MB sd cards and I can confirm that this is a sd card problem.
Format the sd card before copying firmware.bin -> Works always, FIMRWARE.CUR created
Just copy the firmware.bin -> After 3-4 times -> Stops working, FIRMWARE.BIN remains

boelle, 2019-10-12:
so this is a hardware problem and not a firmware problem?

boelle, 2019-10-12:
will close this one as it seems to be related to bad hardware and not related to marlin

we can always reopen

closed 2019-10-12 by boelle

TimoBre, 2019-10-26:
A short feedback.
I find out  that SKR Boards (E3 DIP) like more NTFS formated SD Cards
I changed formating the card betwen Fat32/Fat and NTFS.

But I have also Fat32 SD- Cards wich work fine with my 2nd SKR 1.3 during flashing.
But changing the art of formating the SD Card can maybe a solution for you.

MstrVLT, 2019-11-27:
>  to kill DFU bootloader on my SKR 1.3 while debugging Marlin with jlin
@igorepi i found this https://github.com/Smoothieware/Smoothieware/tree/edge/bootloader 
bootloader for Smoothieware boards same functional 
I don't know this is right for you (