MarlinFirmware/Marlin #15328: CONTROLLER_FAN on SKR_PRO

davidcgu, 2019-09-21:
Good afternoon

I¡m having an issue while trying to enable controller_fan option on Release from 9 Sept with an SKR_PRO.

I have ennabled options related as follows:
- Controller fan.
-Auto fan E0 (single hotened)
-Cabin light.

Have re-definned pins as follows on pins.h:

// Heaters / Fans
//
#define HEATER_0_PIN       PB1  // Heater0
//#define HEATER_1_PIN       PD14 // Heater1
//#define HEATER_2_PIN        PB0  // Heater1
#define HEATER_BED_PIN     PD12 // Hotbed
#define FAN_PIN            PC8  // Fan0
//#define FAN1_PIN           PE5  // Fan1
//#define FAN2_PIN           PE6  // Fan2

 #define CONTROLLER_FAN_PIN  PB0 
 #define CASE_LIGHT_PIN      PE5    
 #define E0_AUTO_FAN_PIN     PE6

I tried also to define disable the desired pins on pins.h and define on config advance but had same result.

When enabling controller fan option, this one only works when part fan ( FAN_PIN            PC8 ) is enabled but if under any other circunstance.

Curiously when part fan is not enabled, booth (part fan (PC8) and controller fan (PB0) are making blinks all time.

If I dissable controller fan on config.adv and reflash then has no problem of blinks anymore on part Fan.

Any clue how to solve this?

Thanks in advance

boelle, 2019-09-24:
think its better to ask on discord

-Marlin Discord server. Join link: https://discord.gg/n5NJ59y

davidcgu, 2019-09-25:
Solved, seems heater pins are not valid for normal PWM used one of the spare on the extension pins and it's wring fine. So issue is not related to MArlin itself.

closed 2019-09-25 by boelle