MarlinFirmware/Marlin #15336: [BUG] SKR 1.3 TMC Connection Error after restarting printing

tobybonhof, 2019-09-22:
### Description

When first uploading marlin to printer i can use tmc 2209 fine. After reboot or reset the printers shows a TMC connection error and sensorless homing will not work anymore.

shitcreek, 2019-09-24:
Please follow the template and provide more information such as your configuration files. But before you do, have you searched through the issues posted here to see if there isn't already a thread opened for this one? For example, could this issue be related to #15012 ?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle