MarlinFirmware/Marlin #15337: [BUG] Printer freezes randomly during print

asko58, 2019-09-22:
Ever since I updated to Marlin 2 and the board SKR 1.3 the printer randomly freezes during some prints. Here are the key facts:

Machine:
- SKR 1.3 using TMC2209 in UART mode
- Marlin 2, tried lots of versions, latest one as of 9/9/2019

Issue:
- random freezes that can not be reproduced:
-> possibly no freeze printing same gcode, freezing at different points during print, no freezes for a number of prints then again many freezes
- when freezing the motors turn off and the head stops moving
- heaters and fans however stay ON!! which is a POSSIBLE FIRE HAZARD
- the drivers and motors are not overheated as they are only slightly warm to the touch
- Marlin stays responsive, the LCD Menu can be used

As Marlin does not notice that it has stopped, I think this could be a severy firmware issue. Can anyone confirm this or suggest a solution?

asko
[config.zip](https://github.com/MarlinFirmware/Marlin/files/3639867/config.zip)

asko58, 2019-09-22:
Ok, things are getting even stranger: I rebooted the printer and re-ran the print to see if it would crash another time at the same point. No my LCD display has got stuck showing the same values since 0.2mm height and the whole machine is no longer responsive to key presses. But it is still PRINTING! ?! I will leave it to its work and watch...

Rdog618, 2019-09-23:
I'm having the exact same issue, its absolutely maddening.

asko58, 2019-09-23:
> I have noticed that I cannot run stealth chop on E0 or I will get continuous under extrusion. 
Stealth chop does not work in combination with linear advance. I have already turned it off, except for the z axis.

> Turning off stealth chop on any other axis induces emf that you can audiably hear ringing from the steppers, the emf regularly causes the motherboard to freeze so I'm pretty obligated to run stealth chop on x y and z.

I don't have these problems. It rather seems the other way round as stealth chop makes noise with my printer.

asko58, 2019-09-23:
UPDATE: It seems to have gotten worse. Tried a different SD-Card and gcode. The printer stopped at 2.1mm height at the first attempt: no more printing moves but heaters ON! This time the LCD showed "Printing..."
[gcode.zip](https://github.com/MarlinFirmware/Marlin/files/3641981/gcode.zip)

lovethat777, 2019-09-24:
Can you give us more information about your setup ? (heater type, power supply, ...)

salacpavel, 2019-09-25:
I am observing same issue with latest build - SKR E3 mini/DIP + TMC2209 drivers in UART.
Sometimes, printer its just UI that becomes unresponsive, but the printer keeps going and after several seconds (10+) the display starts responding again. Sometimes it jsut stalls untill I power cycle it.
Also storing / reading EPROM contents now hangs the printer completely.

asko58, 2019-09-25:
Thanks fo the responses. My setup is the following:

24v 8.5A Power supply
230V heated bed, controlled via pulse width modulated solid state relais
40w 24v heater cartridge embedded in 20x20 heat block
Tevo Nema 17 steppers (no specific type given) @ around 720-780mA rms
SKR 1.3 Board
TMC 2209 stepper drivers connected over UART
BlTouch Smart clone

Haxk20, 2019-09-27:
Exact same issue for me too on SKR 1.3. At first i tought that maybe i connected something wrong or something but TBH im glad im not alone that has this issue. But this sure has to get fixed as it makes the printer basically unusable at bigger prints.

Haxk20, 2019-09-27:
Setup. Stock Tevo Tarantula PRO with SKR 1.3 and TMC2208 in UART mode.

boelle, 2019-10-12:
@asko58 still having issues?

Haxk20, 2019-10-12:
> @asko58 still having issues?

I will try a file that i know freezes my printer tomorrow and will tell you if its still broken on latest marlin.

boelle, 2019-10-12:
remember to rule of a bad sd card etc

Haxk20, 2019-10-12:
> remember to rule of a bad sd card etc

Not a bad SDcard as this happens with USB connection too with SD card out. The main issue here im concerned is the printer stopping for some reason which you can imagine isnt fun on big print.

Haxk20, 2019-10-13:
@boelle Not exactly 100% sure but print just finished without printer freezing. This was 1 Gcode i coudlnt get to print to 100% and i tried like 7 times. So when more people confirm that they dont experience it anymore i guess then its safe to assume its fixed.

boelle, 2019-10-13:
odd, and you did not change anything?

Haxk20, 2019-10-13:
Only updated marlin

Haxk20, 2019-10-13:
But during the first few minutes the screen was OK but then it made all pixels totally black and never came back. I was unable to control the printer but it was still printing

boelle, 2019-10-30:
@asko58 @Haxk20 

have you tested recently? a lot happens in just 2-3 days

Haxk20, 2019-10-31:
Will build a new version of firmware but did a 6 hour print and it went
just fine and finished just fine. But will try with new firmware.

vivian-ng, 2019-10-31:
I have the same symptoms (print freezes but heating of bed and nozzle still continues; print resumes after about a minute or so) on an ESP32 running TMC2130 via I2S stepper stream. I think @felixstorm also faced similar issues. Not sure if our issues are related to this; if so, then the common denominator is 32-bit + TMC driver.

boelle, 2019-11-03:
will close this one as it seems @asko58 do not have issues and @Haxk20 can print fine

we can reopen if several can confirm the issue (and maybe even say what is wrong)

closed 2019-11-03 by boelle

felixstorm, 2019-11-03:
@vivian-ng I do indeed have TMC modules on my ESP32 PCB, but run them standalone without TMC support inside Marlin, so I do not think that our problem is actually related to this.

Haxk20, 2020-03-12:
@boelle I would like to request you reopen this. IDK what caused this but i just literally changed filament to PETG and put the bed to 85C and 245C at hotend. It printed fine for over an hour then it did this again.
First thing that came to mind is if PSU max current isnt the issue but that would reboot the entire printer and not cause SD card to be unplugged virtually.

thinkyhead, 2020-03-14:
@Haxk20 — Start a new issue and be sure to fill out the template. We want to start with a fresh new page.

marhub, 2020-03-25:
Have the same issue - printer prints but UI freezes until I power cycle it.
It usually happens when I pause and unpause print.
Stock Anet A6.
On stock firmware was all ok.

Haxk20, 2020-03-25:
@marhub You have different issue then. Here it is that UI is fine but printer on its own freezes.

OZZYcorexy, 2020-04-03:
Hello there, 
can confirm the issue described by Haxk20 is present in 2.0 bugfix
examplee code was printed well once then i wanted to print it again and it stopped in about middle (4,5hrs ) print. removed residue from plate and started again and faced same different height, different place.  was able to click over the screen but printer was not reacting at all. 
my setup: custom CoreXY, SKR PRO 1.1, TFT24, 230v bed heater on SSR relay, 50W hotend heater. 24V about 400W main power. 
to force printer move again have to power cycle.

Black6spdZ, 2020-04-11:
also seeing same freeze issue with skr mini e3 v1.2 "TMC2209" in latest 2.1-dev

OZZYcorexy, 2020-04-11:
small update on it guys, recently printing only via octoprint and see no issues. 
when issue was captured  i was printing usually from SD or USB on BIQU TFT 24 screen
my conclusion is that one fails rather than motherboard or Marlin itself

Haxk20, 2020-04-11:
> small update on it guys, recently printing only via octoprint and see no issues.
> when issue was captured i was printing usually from SD or USB on BIQU TFT 24 screen
> my conclusion is that one fails rather than motherboard or Marlin itself

At the moment it seems to be voltage drop in PSU that the SKR just cant handle and drops for a split of a second and as a result it drops SDcard which in turn stops print. I have started using octoPrint as well and see no issue with it. No freezes. Done 64 hour print. No freeze. So i think thats a pretty decent test.

marhub, 2020-04-13:
> @marhub You have different issue then. Here it is that UI is fine but printer on its own freezes.

oh, ok- my bad. 
Any existing issue then ? So i don't create duplicate  :)

kain0m, 2020-05-30:
I had this issue. Tried several different versions of Marlin, symptoms were always slightly different. Freeze at random points, heaters on, LCD stuck, with Version 2.0.5.3 the printer rebooted.

Homebuilt Cartesian printer
SKR 1.3
TMC 2208 in UART (Dual Z), stealthchop, Square wave stepping enabled
Linear advance made no difference
BLTouch (freezes also without BLTouch)
Chinese clone RepRap Discount Smart controller
150W 12V PSU

Anyway, since switching SD Card (the SD in the Display, where I store the GCode), I no longer experience the freezing issue (normally, it would freeze every 3-4 hours of printing on average. No freeze for 30+ printing hours since switching SD card). Had been chasing this for half a year now...

minosg, 2020-06-02:
From a similar thread investigating the same issue on another project.

https://github.com/KevinOConnor/klipper/issues/196

This correspondence with TMC is being attached which explains the lock-up issue

>  I think, the reason for disabling of the TMC2208 driver could be
> 
>     a hard stop of the motor in stealthChop (Step frequency goes from a higher value, e.g. > 0.5 RPS to 0)
> 
>     an abrupt change of motor velocity (Step frequency goes from a higher value to a low value within a single step).
> 
> When using stealthChop, please always make sure, that you use velocity ramping. A hard stop will cut away motor back-EMF at once. As stealthChop is a voltage based chopper, it cannot respond to this at once, like spreadCycle. The result is an overcurrent, and the motor driver goes to overcurrent switch off, until it becomes disabled / enabled again.
> 
> To resolve the problem, please use at least a tiny velocity ramping, when hard stopping the motor, e.g. within a few / a few ten microsteps.

To my understanding this issue could be addressed by:
- velocity ramping. ( Not certain how Linear advance will respect that )
- Setting the hybrid threshold to a value, that when the extrude changes direction suddenly it falls to spread cycle.
- Playing with the MINIMUM_STEPPER_POST_DIR_DELAY/MINIMUM_STEPPER_PRE_DIR_DELAY and overriding the driver default of 20nS

Does marlin have a velocity ramping configuration parameter?

OZZYcorexy, 2020-06-03:
> KevinOConnor/klipper#196

well, it doesn't explains why there is no issue when driven via octoprint.
imo driving section is not a problem here issue is random and never
occurred for me after changing to octoprint.
it should be in the area of BIQU TFT 24 screen and it's card reader.

minosg, 2020-06-03:
Apologies, I kinda responded to the wrong ticket. But that driver design feature is still relevant in that case.

Marlin will freeze with heaters on whenever the flow blocks and waits for an external component to return something, Weather that is the LCD screen in auto-levelling, or the TMC drivers. As a result you can have different causes display the same effect.

From my experience Linear advance trigger's the driver lockup way more often than anything else. but one other reason you could be experiencing it, is the condition of the SD card. is SD_CHECK_AND_RETRY enabled?

OZZYcorexy, 2020-06-03:
tried various SD's, brand new one too.
happens randomly.
as i said on octoprint no issues at all already pushed through few kilos
never freezed.
could be screen with issue itself and not software problem