MarlinFirmware/Marlin #15338: [BUG] Compile error with bugfix-2.0 on REARM when enabling Junction Deviation on a Delta configuration

lookiamluke, 2019-09-22:
**Description**
Compile error with bugfix-2.0 when enabling Junction Deviation on a Delta configuration, see error below:

Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\DebugMonitor.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL_SPI.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\MarlinSerial.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\fast_pwm.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\include\digipot_mcp4451_I2C_routines.c.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\main.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\persistent_store_flash.cpp.o
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:38,

                 from Marlin\src\HAL\HAL_LPC1768\HAL.cpp:25:

Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:1028:4: error: #error "Junction deviation is only compatible with Cartesians."

   #error "Junction deviation is only compatible with Cartesians."

    ^~~~~

In file included from Marlin\src\HAL\HAL_LPC1768\../../core/../inc/MarlinConfig.h:38,

                 from Marlin\src\HAL\HAL_LPC1768\../../core/serial.h:24,

                 from Marlin\src\HAL\HAL_LPC1768\DebugMonitor.cpp:26:

Marlin\src\HAL\HAL_LPC1768\../../core/../inc/SanityCheck.h:1028:4: error: #error "Junction deviation is only compatible with Cartesians."

   #error "Junction deviation is only compatible with Cartesians."

    ^~~~~

Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\persistent_store_sdcard.cpp.o
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:38,

                 from Marlin\src\HAL\HAL_LPC1768\HAL_SPI.cpp:51:

Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:1028:4: error: #error "Junction deviation is only compatible with Cartesians."

   #error "Junction deviation is only compatible with Cartesians."

    ^~~~~

Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\timers.cpp.o
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:38,

                 from Marlin\src\HAL\HAL_LPC1768\persistent_store_flash.cpp:44:

Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:1028:4: error: #error "Junction deviation is only compatible with Cartesians."

   #error "Junction deviation is only compatible with Cartesians."

    ^~~~~

In file included from Marlin\src\HAL\HAL_LPC1768\../../sd/../inc/MarlinConfig.h:38,

                 from Marlin\src\HAL\HAL_LPC1768\../../sd/cardreader.h:24,

                 from Marlin\src\HAL\HAL_LPC1768\main.cpp:38:

Marlin\src\HAL\HAL_LPC1768\../../sd/../inc/SanityCheck.h:1028:4: error: #error "Junction deviation is only compatible with Cartesians."

   #error "Junction deviation is only compatible with Cartesians."

    ^~~~~

In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:38,

                 from Marlin\src\HAL\HAL_LPC1768\persistent_store_sdcard.cpp:29:

Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:1028:4: error: #error "Junction deviation is only compatible with Cartesians."

   #error "Junction deviation is only compatible with Cartesians."

    ^~~~~

In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:38,

                 from Marlin\src\HAL\HAL_LPC1768\timers.cpp:31:

Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:1028:4: error: #error "Junction deviation is only compatible with Cartesians."

   #error "Junction deviation is only compatible with Cartesians."

    ^~~~~

Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_hw_spi.cpp.o
*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL_SPI.cpp.o] Error 1

*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\persistent_store_flash.cpp.o] Error 1

*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\DebugMonitor.cpp.o] Error 1

*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL.cpp.o] Error 1

*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\main.cpp.o] Error 1

*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\persistent_store_sdcard.cpp.o] Error 1

*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\timers.cpp.o] Error 1

**Steps to Reproduce**
Use REARM with RAMPS 1.4
Use the example config files for the Kossel Pro (Only enabling junction_deviation)
Enable junction_deviation in configuration.h
Compile
**Expected behavior:** [What you expect to happen]
Compile would succeed.

**Actual behavior:** [What actually happens]
Compile error as shown above

**Additional Information**
I just yesterday updated my Delta firmware with the latest 2.0 bugfix and this error started. I was using junction deviation previously in March of this year without a compile error.

I tried searching the forums to see if anyone else reported similar issues, and the closest I got was this thread:
https://github.com/MarlinFirmware/Marlin/issues/13103

Trying both of the mentioned solutions with the current Bugfix didn't work.  

Thanks

GMagician, 2019-09-22:
> #error "Junction deviation is only compatible with Cartesians."

I think you missed this in above reports

lookiamluke, 2019-09-24:
Hey GMagician, No --- I saw that. Obviously the error had to do with enabling Junction deviation on a delta but I have previously had my delta running Marlin 2.0 with junction deviation --- 

Can anyone confirm that the current bugfix intentionally no longer supports junction deviation on deltas? and why not? 

Thanks

GMagician, 2019-09-24:
Yes it's intentional, please read #15032

boelle, 2019-10-12:
@lookiamluke still having issues?

boelle, 2019-10-24:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-24 by boelle