MarlinFirmware/Marlin #15339: M85 (kill if inactive) will kick in even when printer is awaiting user input due to filament runout.

AndKe, 2019-09-22:
### Description

M85 (kill if inactive) will kick in even when printer is awaiting user input due to filament runout.

### Steps to Reproduce

configure M85 for kill after some inactive time 
run out of filament (trigger sensor)

**Expected behavior:** [What you expect to happen]
I would expect kill timer to be paused during wait for user input.

**Actual behavior:** [What actually happens]
kill timer runs out, job terminated.  

#### Additional Information
-

boelle, 2019-09-23:
please attach your 2 config files, just as the issue template told you to

AndKe, 2019-09-23:
attached :)

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3641085/Configuration.zip)

shitcreek, 2019-10-15:
M85 is functioning as it is intended. `[M85](http://marlinfw.org/docs/gcode/M085.html)`. 
Having the machine wait **indefinitely** for user input would defeat its purpose. It should be set to how long you are willing to have the machine running unattended.

Perhaps as a feature request, maybe an extra parameter could be added to have it pause the inactivity timer while there is a prompt for user input, or specifically have filament runout routine to pause the timer.

AndKe, 2019-10-15:
yes, there should be a acceptable wait time, especially for printing on beds that will release the part if allowed to cool down.

InsanityAutomation, 2019-10-15:
> 
 or specifically have filament runout routine to pause the timer.

Stepper timeout actually has a defer for filament runout explicitly, at least it did last I looked at it. I was planning to add one for the babystepping display at some point as well.

boelle, 2019-10-16:
so the best action to take would be to close this one and open an [FR] instead?

boelle, 2019-11-03:
@InsanityAutomation did you have spare time to test this one?

boelle, 2019-12-10:
will close this one as M85 works as intended, if we need to change the way it works a FR would be in order just like @shitcreek said

closed 2019-12-10 by boelle

shitcreek, 2019-12-10:
I tested the latest build with `ADVANCED_PAUSE_FEATURE` and `PAUSE_PARK_NO_STEPPER_TIMEOUT` enabled and set kill if inactive to 60 seconds (M85 S60), start a print without filament, and watched the clock. 60 seconds after the extruder heater is shut off, the screen goes blank and the heated bed shuts off.
We should expect the printer halted message at the very least.