MarlinFirmware/Marlin #15341: ["SENSORLESS_HOMING requires a TMC stepper driver with StallGuard  Z axes."

NA7KR, 2019-09-22:
I am getting this error "SENSORLESS_HOMING requires a TMC stepper driver with StallGuard  Z axes."

#define X_DRIVER_TYPE TMC2130
#define Y_DRIVER_TYPE TMC2130
#define Z_DRIVER_TYPE TMC2130
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
//#define Z2_DRIVER_TYPE A4988
//#define Z3_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE TMC2130

  #define SENSORLESS_PROBING // StallGuard capable drivers only

  #if EITHER(SENSORLESS_HOMING, SENSORLESS_PROBING)
    // TMC2209: 0...255. TMC2130: -64...63
    #define X_STALL_SENSITIVITY  8
    //#define X2_STALL_SENSITIVITY X_STALL_SENSITIVITY
    #define Y_STALL_SENSITIVITY  8
    //#define Z_STALL_SENSITIVITY  8
    #define SPI_ENDSTOPS              // TMC2130 only
    #define IMPROVE_HOMING_RELIABILITY
  #endif

thinkyhead, 2019-09-24:
Try this change in `Conditionals_post.h`…
```diff
- #if HOMING_Z_WITH_PROBE
+ #if HOMING_Z_WITH_PROBE && DISABLED(SENSORLESS_PROBING)
    #undef Z_STALL_SENSITIVITY
  #endif
```

thinkyhead, 2019-09-24:
Checking over your configs…
You can't mix `BLTOUCH` with `SENSORLESS_PROBING`.

closed 2019-09-24 by thinkyhead

NA7KR, 2019-09-25:
Why does x and y not work have z disabled

Sent from iPhone

thinkyhead, 2019-09-25:
Disable `SENSORLESS_PROBING`. For some reason it is enabled in your config.

Talha909, 2019-12-14:
> Why does x and y not work have z disabled Sent from iPhone
> […](#)

have you find any solution.

Talha909, 2020-03-21:
have you find any solution.

Talha909, 2020-03-21:
> Why does x and y not work have z disabled Sent from iPhone
> […](#)

did you find any luck