MarlinFirmware/Marlin #15343: Weird behaviour / Problem after G29

supagusti, 2019-09-23:
First, thanks for your excellent work!
I've modded my AM8 to use dualX as well as dual Z axis. Using the RAMPS board and 2 stepper extender boards I have 7 TMC 2208 (standallone) stepper drivers in total. 

The issue that I am facing occurs to me in the version 1.1.9 (downloaded 2019-09-18) as well as in the 2.0-bugfix (downloaded 2019-09-19).
I'm not sure, but it could be similar to issue #6636.

When I issue the following G-code commands:
G28
G29
G1 X110 Y110 F4000

the toolhead moves very slowly towards the X endstop and then after it hits the endstop moves (at full speed, only restricted by DEFAULT_MAX_FEEDRATE) to the position in the G1 command.

(-> See video: [Marlin2.0._Error_after_G29.MOV.zip](https://github.com/MarlinFirmware/Marlin/files/3640654/Marlin2.0._Error_after_G29.MOV.zip))

The issue occurs ONLY, when you send the commands, while the printer is moving during G28 and G29  (e.g. type fast in pronterface, or run from SD). When I wait with the last command until G29 is finished. The command G1 X110 Y110 F4000 executes as expected.

I've tried a bltouch (china clone, v. 1.0) and a capacitive probe.

The issue also occurs, when you do a M420 S0 after a G28.
I've tried to put a M400 after the G29 and also a G90 (according to the last post of #6636)

I've also attached my configs, the pins file is also modyfied to fit my needs for 7 steppers: 
[ConfigFiles.zip](https://github.com/MarlinFirmware/Marlin/files/3640653/ConfigFiles.zip)
(I've transferd the values by hand into the 2.0 configs)

Thanks in advance...

boelle, 2019-10-12:
is this still an issue with latest 2.0.x ?

supagusti, 2019-10-17:
I'm a bit confused now. I didn't change anything in the firmware, except for some nozzle offset values. But the issue seems to be gone now. 
I just prepared to upload the latest build and wanted to confirm the issue with my current build, but it's not present anymore.

closed 2019-10-17 by supagusti

jukisu23, 2020-04-06:
I think I'd check in.
I had the same exact Problem.
After some time I realized the following:
Cura automatically adds the nozzle lifting in the end of the print. For this it issues **G91** for relative positioning.
If you don't manually add a **G90** (absolute positioning) to the start gcode, it will still be in relative mode and therefore always be stuck in the back right corner.

thinkyhead, 2020-04-08:
It has always been a best practice to put `G90` at the top of your G-code because you never know what mode the machine will be in. Some printers may even default to relative mode, since that is a configuration option.