MarlinFirmware/Marlin #15346: [FR] HOST_PROMPT_SUPPORT

maukcc, 2019-09-23:
When using HOST_PROMPT_SUPPORT for filament out, we get a prompt stating the filament is out, without further buttons to do anything.

when we look in host_actions.cpp we see 3 buttons that should be generated. 
Is there a reason that these are not shown?

It would be helpfull if we (peolpe that do not code for a living) would be able to make these prompts to our liking.

InsanityAutomation, 2019-09-23:
IIRC the initial prompt stating its out is informational, while the parking action takes place then once its ready for input the prompt is swapped. It cannot be modified as the responses must mirror what actions the graphical LCD takes and the blocking dialogs on the machine expect.

maukcc, 2019-09-23:
Thanks for the info.

closed 2019-09-23 by maukcc