MarlinFirmware/Marlin #15348: [FR] Standard protocol for communication with "smart" displays

extesy, 2019-09-24:
Some 3d printer manufacturers (Wanhao, Creality, etc) use "smart" colored touch screens which are capable of showing real-time print information such as temperature, time estimates, etc. They use modified Marlin for that and don't release their source code so nobody can reproduce the same functionality with other displays or motherboards.

So I'd like to request Marlin dev team to come up with some kind of a standard communication protocol with such display panels and let all Marlin users enjoy their beautiful displays and not have to fall back to ugly LCD12864 mode.

jorgeassuncao, 2019-09-24:
+1 for that!

vMeph, 2019-09-24:
this would be awesome !!!

grego1981, 2019-10-04:
+1 from me!!

extesy, 2020-06-30:
Looks like there's still demand, please don't close the ticket.

sjasonsmith, 2020-11-10:
@InsanityAutomation isn't this what ExtUI is for? Is there actually anything actionable here, or is it done?

InsanityAutomation, 2020-11-10:
@sjasonsmith pretty much, but a standardized protocol as mentioned is impossible as every screen needs a different protocol. ExtUI is as close as this is going to get.

closed 2020-11-10 by InsanityAutomation