MarlinFirmware/Marlin #15350: [QUESTION] how to know if homing was done ?

spilz87, 2019-09-24:
Hello,

I’m looking for a gcode that can tell me if M114 returned values are “good”.
I mean if homing has been done before. 
It’s same kind of information than when “???” Blink or not on XYZ value on screen but through gcode 

Does it exist ?

Thanks for your help

asko58, 2019-09-24:
You could issue a 'G28 O', if the position is known it will just do nothing, otherwise it will home.

spilz87, 2019-09-24:
Thanks for your reply
But it’s to monitor a printer, so I don’t want to run a home, just get information :(

boelle, 2019-09-24:
i think its better to ask on discord

-Marlin Discord server. Join link: https://discord.gg/n5NJ59y

han9988, 2019-10-08:
You could check the state of your endstops with M119. For example:

Sucessful homing:
 x_min: TRIGGERED
y_min: TRIGGERED
z_min: TRIGGERED

Ooopsy - something went wrong:
x_min: TRIGGERED
y_min: open
z_min: TRIGGERED

boelle, 2019-10-24:
better ask here: 

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

the list is for bugs and fr's only

closed 2019-10-24 by boelle