MarlinFirmware/Marlin #15351: [BUG] Strange arc (G2/G3) movement

MasterTim17, 2019-09-24:
Hello,

### Description
i want to laser cut a text on paper. Im using the inkscape programm to generate a text and generate a gcode with the jtech photonic laser plugin. The acceleration is set to 100 with M201. The outputted gcode is looking correct (verified in gcode viewer).

My printer is using Marlin Firmware 2.0 bugfix on a SKR V1.3 with TMC2208 stepper motor drivers. ARC_SUPPORT is enabled in the firmware.

**Expected behavior:** [What you expect to happen]
Correct arc movement to draw the text

**Actual behavior:** [What actually happens]
Weird lines

[image]

boelle, 2019-09-24:
please attach the 2 config files just as the issue template told you to

DerAndere1, 2019-09-24:
@MasterTim17 In case you know how to write G-code manually, can you provide a minimal G-code example (best would be < 5 lines including one G2/G3 command) and compare expected vs observed movement? from the attached picture I can only guess that there are many G2 commands affected. This is likely caused by something else than the rounding issue https://github.com/MarlinFirmware/Marlin/issues/14745 .

MasterTim17, 2019-09-24:
Here is a simple Gcode and the resulted printed movement
[Arc.zip](https://github.com/MarlinFirmware/Marlin/files/3647807/Arc.zip)

DerAndere1, 2019-09-24:
The G2 commands are obviously ignored: It does not move when G2 commands are issued. G1 commands are executed as expected. How do you send the G-code to your hardware? Does it help to use SD card or octoprint or printrun? Also, does it help to round values to 3 decimals? see https://github.com/kliment/Printrun/issues/620#issuecomment-74273532

MasterTim17, 2019-09-24:
i just looked in my terminal of octoprint: there are only g1 commands.
Then i found this solution: the plugin exclude region blocks g2/g3 commands 
[Octoprint not sending G2/G3](https://community.octoprint.org/t/octoprint-not-sending-g2-g3/5645)

DerAndere1, 2019-09-24:
Cool. If you have no more issues with G2/G3 you can close this issue. Regards.

closed 2019-09-24 by MasterTim17