MarlinFirmware/Marlin #15352: How to scroll 6 heating bar ICONS on LCD12864

yangwenxiong, 2019-09-24:
### Description

On the LCD12864 screen, if all the heating rods are opened in the program, only four heating rod ICONS can be displayed.

Excuse me,How to scroll 6 heating bar ICONS on LCD12864?

#### Additional Information
[image]

[image]

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

boelle, 2019-09-24:
please try the following resources:

-RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
-[Tom's 3D Forums](https://forum.toms3d.org/)
-Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
-Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
-[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
-Marlin Discord server. Join link: https://discord.gg/n5NJ59y

shitcreek, 2019-09-24:
Look inside `dogm_Statusscreen.h` and you will see that there are only drawings for up to 4 extruders. You will need to add your own custom condition for 6 extruders. You will likely need to make a special case that would nudge them over and tighter to fit all 6. I am currently working on a minimalist interface that would attempt to fit everything onto the screen.

Closing because this is not a bug. Please open a Feature Request if you feel inclined to.

closed 2019-09-24 by shitcreek