MarlinFirmware/Marlin #15353: [BUG] skr 1.3 12864 LCD reboot loop after more than 15 files

wolfgangmauer, 2019-09-24:
### Description
I have a SKR 1.3 and a 12864 LCD-Grafik-Smart-Display-Controller
If i have more than 15 files on the sdcard in the LCD marlin won't boot (boot/reset/boot/reset/...)

### Steps to Reproduce
Put more than 15 files on the sdcard an insert in the 12864 LCD and reboot
After that -> reboot loop

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3646559/Configuration.zip)

shitcreek, 2019-09-24:
Does the SD card contain a .cur, .bin or an auto0.g file?

wolfgangmauer, 2019-09-24:
No, just gcode files.

If i delete some to get less than 16 everything works fine

shitcreek, 2019-09-25:
I have well over 16 gcode files on my sd card without this issue.
Have you tried a different SD card?

gloomyandy, 2019-09-25:
I suspect that you are seeing a variant on this issue https://github.com/MarlinFirmware/Marlin/issues/14326
I think that you will be getting a watchdog reset during boot. You could either try the fix I suggested or try disabling: USE_WATCHDOG in configuration_adv.h.NOTE you probably should not run your printer with the watchdog disabled. I am suggesting this only to help track down what is going on.

The reason you see this but others may not is probably because something in your boot sequence is taking a little longer than other systems to run (or you may have other things enabled that others do not have). This extra delay means that the watchdog timer expires on your system, but not on others.

wolfgangmauer, 2019-09-25:
THX!

Yes i have the „SDCARD_SORT_ALPHA“ enabled…

I will check without

wolfgangmauer, 2019-09-25:
Disable "SDCARD_SORT_ALPHA" fix this issue

closed 2019-09-25 by wolfgangmauer

firepower9966, 2019-09-27:
You need to make sure you have sort use ram or stack enabled, I got same issue with mega2560 when I disable ram defines in sort alpha.  Also sorting alpha is limited to 40 entries I think.

thinkyhead, 2019-09-27:
> sorting alpha is limited to 40 entries I think

The limit is 255. A Melzi with 16K of SRAM can handle this with no problem.