MarlinFirmware/Marlin #15355: [BUG] ?? SKR PRO NEOPIXEL WRONG COLOURS

SISLANGER, 2019-09-24:
### Description

#define NEOPIXEL_TYPE   NEO_GRB works on SKR V1.3
 but wrong colours on SKR PRO V1.1

**Expected behavior:** 
Correct Colour

**Actual behavior:** [What actually happens]
Startup is wrong 

Colours at moment
Red produces green
Green produces blue
Blue produces red

boelle, 2019-09-24:
wires connected correct on SKR PRO V1.1?

SISLANGER, 2019-09-24:
Yeah signal to signal 
gnd to gnd

gnd & 5V off buck convertor

SISLANGER, 2019-09-24:
also now will no longer light more than 2 - 4 neopixels despite 77 set

extesy, 2019-09-24:
@SISLANGER I think you should ask @bigtreetech in https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/ first before involving Marlin.

boelle, 2019-09-24:
i would agree, if its a hardware issue then seek help from the maker of hardware :-)

SISLANGER, 2019-09-24:
If it has been implemented into Marlin then it would be seen by author of that coding. Well at least that is my take on it. Sadly I can't just message them. I don't believe it to be hardware. Seems more of a timing issue with neopixel. Or perhaps Marlin 2 not updating firmware with the SKR Pro V1.1 like I experienced with SKR V1.3

If you wish to close the case then that is up to you

sjasonsmith, 2019-09-26:
@SISLANGER, can you provide a minimal config that can reproduce the problem, that I can load and use on a board with nothing else connected?

I don't have a NeoPixel yet, but I'm expecting one next Wednesday. I could try to reproduce the issue. I would need the config and exact steps to reproduce (I've never used LEDs on my printers before).

Promite2501, 2019-09-29:
Are you running a split ground to both the skr and buck converter?

SISLANGER, 2019-09-30:
@sjasonsmith 

In simplest form

// Support for Adafruit Neopixel LED driver
#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE    NEO_GRB // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
  #define NEOPIXEL_PIN     PC9 //PB0 //PC9 // PE6 // LED driving pin
  //#define NEOPIXEL2_TYPE NEOPIXEL_TYPE
  //#define NEOPIXEL2_PIN    5
  #define NEOPIXEL_PIXELS 77 //77       // Number of LEDs in the strip, larger of 2 strips if 2 neopixel strips are used
  //#define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
  #define NEOPIXEL_BRIGHTNESS 255  // Initial brightness (0-255)
  #define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup

  // Use a single Neopixel LED for static (background) lighting
  //#define NEOPIXEL_BKGD_LED_INDEX  0               // Index of the LED to use
  //#define NEOPIXEL_BKGD_COLOR { 255, 255, 255, 0 } // R, G, B, W
#endif

#if ANY(BLINKM, RGB_LED, RGBW_LED, PCA9632, PCA9533, NEOPIXEL_LED)
  #define PRINTER_EVENT_LEDS
#endif

#define NUM_SERVOS 1 // Servo index starts with 0 for M280 command
#define SERVO_DELAY {300}

SISLANGER, 2019-09-30:
@Promite2501 I have ground from PSU and signal meeting together soldered to lights together, same as always.

Evg33, 2019-10-02:
@SISLANGER Play with #define NEOPIXEL_TYPE NEO_GRB 
May be NEO_RBG ?
Other Varants:
NEO_RGB
NEO_GRB
NEO_GBR
NEO_BRG
NEO_BGR

sjasonsmith, 2019-10-03:
I just set up an authentic 12-pixel Adafruit NeoPixel RGB ring (model 1643) on an SKR Pro 1.1, and the colors are fine. I see the startup sequence at boot, then I manually tested colors using the M150 command.

I used the same pin from the settings @SISLANGER posted.
Here are all the enabled settings in my NeoPixel section:

```
// Support for Adafruit Neopixel LED driver
#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE   NEO_GRB
  #define NEOPIXEL_PIN     PC9
  #define NEOPIXEL_PIXELS 12
  #define NEOPIXEL_IS_SEQUENTIAL
  #define NEOPIXEL_BRIGHTNESS 127  // Initial brightness (0-255)
  #define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup
#endif
```

Here are my configs if you want to compare. This a nearly completely clean config, just essential board and NeoPixel settings modified.
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/3684323/Configs.zip)

This was built on the latest Oct 2nd bugfix, commit b82c0978cc38f429377af25d64a57408b802cd99. The Neopixel library is version 1.2.5.

Evg33, 2019-10-03:
I have no problems with colors.
connected 2 different tapes (RGB and GRB) to different outputs.

Configuration.h
```
  #define NEOPIXEL_TYPE   NEO_RGB // RGB 
  #define NEOPIXEL_PIN     P1_21  // LED driving pin
  #define NEOPIXEL2_TYPE NEO_GRB  // GRB
  #define NEOPIXEL2_PIN P2_00     // Servo connector
```

pins_BIGTREE_SKR_V1.3.h:
`#define SERVO0_PIN         -1 //P2_00
`

SISLANGER, 2019-10-03:
> @SISLANGER Play with #define NEOPIXEL_TYPE NEO_GRB
> May be NEO_RBG ?
> Other Varants:
> NEO_RGB
> NEO_GRB
> NEO_GBR
> NEO_BRG
> NEO_BGR

Yeah I have tried those, and I have the colour working correctly but only a few LEDS

SISLANGER, 2019-10-03:
> I just set up an authentic 12-pixel Adafruit NeoPixel RGB ring (model 1643) on an SKR Pro 1.1, and the colors are fine. I see the startup sequence at boot, then I manually tested colors using the M150 command.
> 
> I used the same pin from the settings @SISLANGER posted.
> Here are all the enabled settings in my NeoPixel section:
> 
> ```
> // Support for Adafruit Neopixel LED driver
> #define NEOPIXEL_LED
> #if ENABLED(NEOPIXEL_LED)
>   #define NEOPIXEL_TYPE   NEO_GRB
>   #define NEOPIXEL_PIN     PC9
>   #define NEOPIXEL_PIXELS 12
>   #define NEOPIXEL_IS_SEQUENTIAL
>   #define NEOPIXEL_BRIGHTNESS 127  // Initial brightness (0-255)
>   #define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup
> #endif
> ```
> 
> Here are my configs if you want to compare. This a nearly completely clean config, just essential board and NeoPixel settings modified.
> [Configs.zip](https://github.com/MarlinFirmware/Marlin/files/3684323/Configs.zip)
> 
> This was built on the latest Oct 2nd bugfix, commit [b82c097](https://github.com/MarlinFirmware/Marlin/commit/b82c0978cc38f429377af25d64a57408b802cd99). The Neopixel library is version 1.2.5.

I will give yesterdays update a try, and check neopixel version. Thank you

Evg33, 2019-10-03:
@SISLANGER 
> Yeah I have tried those, and I have the colour working correctly but only a few LEDS
Can you show the photo of strip with M150 commands ?
Maybe you have 4 color strip ? (NEO_RGBW...)

boelle, 2019-10-12:
@SISLANGER

Can you show the photo of strip with M150 commands ?
Maybe you have 4 color strip ? (NEO_RGBW...)

boelle, 2019-10-24:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-24 by boelle