MarlinFirmware/Marlin #15357: Tevo Tornado example #define ARC_SUPPORT save memory

Spartacus1066, 2019-09-24:
Not a big deal, but unsure why the example has #define ARC_SUPPORT in Configuration_adv.h
There is no slicer I know of that will produce G2/G3 for the Tevo Tornado. ~3k of memory.

boelle, 2019-09-24:
V1 or V2?

shitcreek, 2019-09-24:
`ARC_SUPPORT` are more for CNC engraving and milling I'd say. A G2/G3 command will greatly reduce the size of the gcode file.
Since Marlin was build upon GRBL which was made for CNCing, it was probably kept for that reason. I would not be opposed to having this feature disabled by default in the vanilla Marlin. Thanks for bringing it up. I will bring it this up when we go over cleaning up the configuration files.

closed 2019-09-24 by shitcreek