MarlinFirmware/Marlin #15362: Changing EXTRUDER_AUTO_FAN_SPEED to something else but 255 makes the build fail unexpectedly

salacpavel, 2019-09-25:
## Description

I am building fresh pull (today) of Marlin bugfix-2.0.x for Fysetc Cheetah 1.2 board. According to mainboard datasheet and as per its schematic, this board has FAN_1 (extruder cooling fan) pin attached to ST32F1 pin PB0, which is supposedly PWM capable. PB0 pin is correctly assigned in pins_FYSETC_CHEETAH_V12.h definition file as FAN1_PIN.
As I want to use extruder AUTO_FAN feature, I have  #define E0_AUTO_FAN_PIN FAN1_PIN. On its own, this does not cause any issues and build compiles/work correctly.
However, as soon as I change #define EXTRUDER_AUTO_FAN_SPEED to anything else but 255, buld fails with following error:

In file included from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:38:0,
                 from Marlin\src\HAL\HAL_STM32F1\HAL.cpp:31:
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:1848:5: error: non-constant condition for static assertion
     static_assert(PWM_PIN(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
     ^~~~~~~~~~~~~
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:1848:5: error: the value of 'PIN_MAP' is not usable in a constant expression
In file included from C:\users\paa\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:54:0,
                 from C:\users\paa\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
                 from Marlin\src\HAL\HAL_STM32F1\../shared/Marduino.h:36,
                 from Marlin\src\HAL\HAL_STM32F1\HAL.h:32,
                 from Marlin\src\HAL\HAL_STM32F1\HAL.cpp:30:
C:\users\paa\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/boards.h:59:29: note: 'PIN_MAP' was not declared 'constexpr'
 extern const stm32_pin_info PIN_MAP[];
                             ^~~~~~~
In file included from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:38:0,
                 from Marlin\src\HAL\HAL_STM32F1\HAL_SPI.cpp:35:
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:1848:5: error: non-constant condition for static assertion
     static_assert(PWM_PIN(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
     ^~~~~~~~~~~~~
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:1848:5: error: the value of 'PIN_MAP' is not usable in a constant expression
In file included from C:\users\paa\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:54:0,
                 from C:\users\paa\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
                 from d:\marlin-bugfix-2.0.x\marlin\src\hal\shared\marduino.h:36,
                 from d:\marlin-bugfix-2.0.x\marlin\src\hal\hal_stm32f1\hal.h:32,
                 from d:\marlin-bugfix-2.0.x\marlin\src\hal\hal.h:26,
                 from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\HAL_STM32F1\HAL_SPI.cpp:35:
C:\users\paa\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/boards.h:59:29: note: 'PIN_MAP' was not declared 'constexpr'
 extern const stm32_pin_info PIN_MAP[];
                             ^~~~~~~
In file included from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:38:0,
                 from Marlin\src\HAL\HAL_STM32F1\Servo.cpp:26:
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:1848:5: error: non-constant condition for static assertion
     static_assert(PWM_PIN(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
     ^~~~~~~~~~~~~
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:1848:5: error: the value of 'PIN_MAP' is not usable in a constant expression
In file included from C:\users\paa\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:54:0,
                 from C:\users\paa\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
                 from d:\marlin-bugfix-2.0.x\marlin\src\hal\shared\marduino.h:36,
                 from d:\marlin-bugfix-2.0.x\marlin\src\hal\hal_stm32f1\hal.h:32,
                 from d:\marlin-bugfix-2.0.x\marlin\src\hal\hal.h:26,
                 from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\HAL_STM32F1\Servo.cpp:26:
C:\users\paa\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/boards.h:59:29: note: 'PIN_MAP' was not declared 'constexpr'
 extern const stm32_pin_info PIN_MAP[];
                             ^~~~~~~
In file included from d:\marlin-bugfix-2.0.x\marlin\src\inc\marlinconfig.h:38:0,
                 from Marlin\src\HAL\HAL_STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp:29:
d:\marlin-bugfix-2.0.x\marlin\src\inc\SanityCheck.h:1848:5: error: non-constant condition for static assertion
     static_assert(PWM_PIN(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
     ^~~~~~~~~~~~~
d:\marlin-bugfix-2.0.x\marlin\src\inc\SanityCheck.h:1848:5: error: the value of 'PIN_MAP' is not usable in a constant expression
In file included from C:\users\paa\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:54:0,
                 from C:\users\paa\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
                 from d:\marlin-bugfix-2.0.x\marlin\src\hal\shared\marduino.h:36,
                 from d:\marlin-bugfix-2.0.x\marlin\src\hal\hal_stm32f1\hal.h:32,
                 from d:\marlin-bugfix-2.0.x\marlin\src\hal\hal.h:26,
                 from d:\marlin-bugfix-2.0.x\marlin\src\inc\marlinconfig.h:30,
                 from Marlin\src\HAL\HAL_STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp:29:
C:\users\paa\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/boards.h:59:29: note: 'PIN_MAP' was not declared 'constexpr'
 extern const stm32_pin_info PIN_MAP[];
                             ^~~~~~~
*** [.pio\build\STM32F103R_fysetc\src\src\HAL\HAL_STM32F1\HAL.cpp.o] Error 1
*** [.pio\build\STM32F103R_fysetc\src\src\HAL\HAL_STM32F1\HAL_SPI.cpp.o] Error 1
*** [.pio\build\STM32F103R_fysetc\src\src\HAL\HAL_STM32F1\Servo.cpp.o] Error 1
*** [.pio\build\STM32F103R_fysetc\src\src\HAL\HAL_STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp.o] Error 1
================================================================= [FAILED] Took 6.39 seconds ================================================================= 

### Steps to Reproduce

1. Setup build environment with attached config files
2. Start build process

**Expected behavior:** 
Build should either finish green, or (if PB0 is in fact not a PWM capable pin) build process should stop with following message instead "_AUTO_FAN_PIN is not a PWM pin. Set EXTRUDER_AUTO_FAN_SPEED to 255."

**Actual behavior:** 
Fails with above error message

#### Additional Information

[config.zip](https://github.com/MarlinFirmware/Marlin/files/3651595/config.zip)

salacpavel, 2019-09-30:
By the way this bug can also be reproduced on build from Aug.8th, which I happen to have archived.

thinkyhead, 2019-10-08:
We're aware of the issue. The sanity-check wants to know if the pin can do PWM, but the HAL definition for this uses a run-time function. This issue will be fixed once we update all the HALs to be able to answer this question at compile-time.

@p3p — What are the chances we can get STM32 to respond to "is this a PWM pin?" statically? Are pins PWM-able or non-PWM-able based on something at runtime?

SwiftNick, 2019-10-23:
There does seem to be a limitation with autofan here, as the extruder fan can be controlled with GCODE as a second fan using a soft pwm pin. But you can't assign this same pin as an autofan unless it is either fully on or off.

LinoBarreca, 2019-11-11:
@thinkyhead @p3p As far as I know there's no way to answer at compile time since on STM32 every pin COULD be PWM if defined in the periphalpins PinMap_PWM[] array.
I might be wrong but without a loop there's no way to see if that array contains the pin unless you write a very very long #define checking every element, if exists...basically an #if 100 pins long...The proper approach would be to use that static assert only on those platforms which have a constant PWM pin list.

boelle, 2019-12-07:
@salacpavel since 2.0 was just released a few days ago has this changed this issue at all?

salacpavel, 2019-12-07:
I do not know, sorry. Has it? IMO devs should give an answer here. I mean this is what I would expect to be the case.

boelle, 2019-12-07:
> I do not know, sorry. Has it?

that is why i ask, i'm not a dev 

and there are only a few devs and they all work for free so they have limited time to spend on marlin, the best thing users can do is to test

salacpavel, 2019-12-07:
I understand. But this seems to be common problem here. Issue gets reported, verified and of course we wait and hope that someone fixes them. (Yes I know, its not granted.) But what is the point when no developer responds and eventually, issue gets closed as "being too old", without being resolved in fact?
Very flustrating.

boelle, 2019-12-07:
> But what is the point when no developer responds and eventually, issue gets closed as "being too old", without being resolved in fact?

i try to only close issues where the reporter stops reponding

but also remember that no dev is forced to do anything, marlin is not a paid product so you cant demand anything from it

Forsvn, 2019-12-28:
Cheetah 32bit v1.2 same problem with changing fan speed, did someone solve the problem?
#define AF_ERR_SUFF "_AUTO_FAN_PIN is not a PWM pin. Set EXTRUDER_AUTO_FAN_SPEED to 255."

STM32  FYSETC_Cheetah 32bit v1.2

Forsvn, 2019-12-31:
I solved the question of how to go through compilation (two ways, but how will it be correct and is it correct at all?).
I set the fan to turn on at a temperature of 50 degrees, the speed is 178 - we get a compilation error.
in the file SanityCheck.h we find:
#if HAS_AUTO_FAN && EXTRUDER_AUTO_FAN_SPEED! = 255
  #define AF_ERR_SUFF "_AUTO_FAN_PIN is not a PWM pin. Set EXTRUDER_AUTO_FAN_SPEED to 255."
and we correct, I have compilation in two cases - which is more correct for the printer ?:
1.) #if HAS_AUTO_FAN && EXTRUDER_AUTO_FAN_SPEED! = 178
  #define AF_ERR_SUFF "_AUTO_FAN_PIN is not a PWM pin. Set EXTRUDER_AUTO_FAN_SPEED to 178."

2.) #if HAS_AUTO_FAN && EXTRUDER_AUTO_FAN_SPEED! = 178
  // # define AF_ERR_SUFF "_AUTO_FAN_PIN is not a PWM pin. Set EXTRUDER_AUTO_FAN_SPEED to 255."

and will this solve this problem?

boelle, 2020-01-24:
@salacpavel when you get time please have a retest

and as bugfix 2.0.x is updated daily please retest say every 14 days or so

boelle, 2020-01-30:
issue also noted here: https://github.com/MarlinFirmware/Marlin/issues/16721

salacpavel, 2020-01-31:
2.0.3:

boelle, 2020-01-31:
and with bugfix 2.0.x ?

zalex94, 2020-02-03:
Globally same behavior with marlin bugfix 2.0.x with SKR PRO 1.1 board.
[Bugfix_Error.txt](https://github.com/MarlinFirmware/Marlin/files/4149595/Bugfix_Error.txt)

kkosa1, 2020-03-01:
Problem still exists in 2.0.4.4 release as of download on 02-28-20 @ 10:00 am with SKR Pro V1.1

kkosa1, 2020-03-04:
I have been watching the site and find myself wondering if this issue might be considered for resolution.  possibly it has been dealt with and i have missed it.  

Thank you in advance for any response.

kingofmonkeys, 2020-03-11:
Just wanted to add to the list of people who hit this error and would like to see it fixed.  I'm running a BTT SKR PRO.

closed 2020-03-29 by thinkyhead

kkosa1, 2020-03-30:
I saw this Merged and Closed this evening and hoped this feature was now available.

Downloaded bugfix 1/2 hour ago.  
with  #define EXTRUDER_AUTO_FAN_SPEED 200 
I can not compile,
error: non-constant condition for static assertion 

all is good with default of 255, please advise

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4400273/Configuration.zip)

BlackMelt, 2020-04-03:
@thinkyhead this issue is still present on my Machine (bugfix 2.0.x , downloaded 02.04.2020 18:30). 🤷‍♂️

PrintSeb, 2020-04-19:
Hello, 
i have the same issue on SKR 1.1 pro.
you can comment out in Marlin\src\inc\SanityCheck.h 
(line 1993)static_assert(PWM_PIN(E0_AUTO_FAN_PIN),... -> this fix the compiler error on STM32 boards.

kkosa1, 2020-04-19:
Thank you!  Works now as I had hoped.