MarlinFirmware/Marlin #15363: [FR] Remap STM32F1 JTAG pins to UART

salacpavel, 2019-09-25:
There are now many boards based on STM32F1 MPU, available on the market. SM32F1 is a very low cost 32bit solution and I believe it will gain even more popularity among makers/manufacturers eventually. However, boards based on this MPU, generaly only have few free pins for external sensors, due to limited MCU pin count.
However, all of these boards have JTAG pinheader, that carries 3.3V, GND and two serial comm pins. 
Would it be possible to remap these pins in Marlin to i.e. general UART RX/TX pins, so that they can be used to connect ESP32 module, running ESP3D web interface?

tpruvot, 2019-09-26:
On the F1, its generally SWD (stlink), not JTAG, and these pins are dedicated to that... to upload and debug SWDCLK and SWDIO... 

They can in theory be remapped to PA14/PA13 but they are in this case normal gpios, not hardware serial

p1ngb4ck, 2019-11-15:
Even 2 additional GPIOs would help a lot on a board like the SKR 1.3

ETE-Design, 2019-11-18:
Will it be possible to do on the MKS robin Lite 1.1? Schematic: https://github.com/makerbase-mks/MKS-Robin/blob/master/MKS%20Robin%20Lite/hardware/MKS%20Robin%20Lite%20V1.1_002/MKS%20Robin%20Lite%20V1.1_002%20SCH.pdf

And will it be possible it that way to add BLTouch?

closed 2020-06-30 by github-actions[bot]