MarlinFirmware/Marlin #15364: [BUG] DON'T COMPILE WITH (BL)3DTOUCH

Cyberhussar, 2019-09-25:
[image]
 
It doesn't work with M851.cpp fix.

Bergerac56, 2019-09-25:
I confirm this issue

This is not related to BL touch. I have the same issue with another type of Z Probe

reloxx13, 2019-09-25:
link fix https://github.com/MarlinFirmware/Marlin/commit/b4ea1875bf6cc24fe4cd47967342d87116f329b3

closed 2019-09-25 by thinkyhead

Cyberhussar, 2019-09-25:
[image]

Still doesn't work :( I changed 44th comand line, saved and rebuilded with C, after all i have the same issue.

sl1pkn07, 2019-09-25:
works for me

Cyberhussar, 2019-09-26:
For me too.

Wysłane z iPhone'a