MarlinFirmware/Marlin #15365: [BUG]  Last bugfix version  does not compile

Bergerac56, 2019-09-25:
Last version  da1bda493e0c7472673821e18d039a2ba13200ae   does not compile for me (MKS SBASE)

It seems to have been introduced by commit df1e51258a8bcc432deed9f5bd60df41a75b2c32

Error message:

Compiling .pio\build\LPC1768\src\src\gcode\queue.cpp.o
Marlin\src\gcode\probe\M851.cpp:38:120: error: macro "SERIAL_ECHO_MSG" passed 6 arguments, but takes just 1
     SERIAL_ECHO_MSG(MSG_PROBE_OFFSET " X", probe_offset[X_AXIS], " Y", probe_offset[Y_AXIS], " Z", probe_offset[Z_AXIS]);
                                                                                                                        ^
Compiling .pio\build\LPC1768\src\src\gcode\scara\M360-M364.cpp.o
Marlin\src\gcode\probe\M851.cpp: In static member function 'static void GcodeSuite::M851()':
Marlin\src\gcode\probe\M851.cpp:38:5: error: 'SERIAL_ECHO_MSG' was not declared in this scope
     SERIAL_ECHO_MSG(MSG_PROBE_OFFSET " X", probe_offset[X_AXIS], " Y", probe_offset[Y_AXIS], " Z", probe_offset[Z_AXIS]);
     ^~~~~~~~~~~~~~~
Marlin\src\gcode\probe\M851.cpp:52:12: error: assignment of read-only variable 'ok'
       ok = false;
            ^~~~~
Marlin\src\gcode\probe\M851.cpp:62:12: error: assignment of read-only variable 'ok'
       ok = false;
            ^~~~~
Marlin\src\gcode\probe\M851.cpp:72:12: error: assignment of read-only variable 'ok'
       ok = false;
            ^~~~~
*** [.pio\build\LPC1768\src\src\gcode\probe\M851.cpp.o] Error 1

closed 2019-09-25 by Bergerac56

Cyberhussar, 2019-09-25:
[image]
 I think i have the same problem :( 

3D touch ? If you switch off ABL it will run :)

alienboyxp, 2019-09-30:
@Bergerac56 how you fix this issue? I need to compile with BLTOUCH for SKR 1.3.
Maybe was fixed in a new commit?
Thanks.

InsanityAutomation, 2019-09-30:
@alienboyxp yup, just grab the latest.

Bergerac56, 2019-10-05:
I confirm ;)