MarlinFirmware/Marlin #15368: Compile error [BUG] (short description)

Glod76, 2019-09-25:
### Description

In file included from Marlin\src\lcd\extensible_ui\lib\lulzbot\language\language.h:87,
                 from Marlin\src\lcd\extensible_ui\lib\lulzbot\language\language.cpp:25:
Marlin\src\lcd\extensible_ui\lib\lulzbot\language\language_en.h:69:49: error: expected unqualified-id before '=' token
   PROGMEM Language_Str JUNCTION_DEVIATION       = u8"Junc Dev";
                                                 ^
In file included from Marlin\src\lcd\extensible_ui\lib\lulzbot\language\language.h:89,
                 from Marlin\src\lcd\extensible_ui\lib\lulzbot\language\language.cpp:25:
Marlin\src\lcd\extensible_ui\lib\lulzbot\language\language_fr.h:51:49: error: expected unqualified-id before '=' token
   PROGMEM Language_Str JUNCTION_DEVIATION       = u8"D├⌐viat. jonct.";
                                                 ^
*** [.pio\build\LPC1768\src\src\lcd\extensible_ui\lib\lulzbot\language\language.cpp.o] Error 1
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3653437/Marlin.zip)

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

sjasonsmith, 2019-09-25:
It looks like with the recent localization improvements by @marcio-ao (#15249), JUNCTION_DEVIATION ended up used as both as the name of a string and a setting in Configuration.h.

I wonder if the localization strings could use a common prefix/suffix (or just not all-caps) to prevent future collisions?

marcio-ao, 2019-09-25:
I am actually working on a PR that will put all the language strings in a namespace and get rid of this `#define` madness. But it will be a while until that is done :)

sjasonsmith, 2019-09-25:
Great! For anyone hitting this, the workaround for right now is to rename JUNCTION_DEVIATION in the language files, and anywhere you find `GET_TEXTF(JUNCTION_DEVIATION))`.

sjasonsmith, 2019-09-25:
This should be fixed with thinkyheads commit [ddd9d9032aa5cf0ee1fea8a17b14428e622cad14](https://github.com/MarlinFirmware/Marlin/commit/ddd9d9032aa5cf0ee1fea8a17b14428e622cad14)

InsanityAutomation, 2019-09-28:
FYI PR renaming these vars to avoid the issue is done and merged. This can be closed.

closed 2019-10-12 by boelle