MarlinFirmware/Marlin #15370: Erratic Servo Deployed Z-Probe Operation

Reywas62, 2019-09-25:
### Description

Probe deploys/retracts unexpectedly during probing sequence.  No heaters or fans active.  SKR V1.3.  I have been able to reproduce the behavior easily using M48 E, never got more than 5 probing cycles before erratic retract occurred and probing failed.   I do not see this issue with a previous pull dated 6//22/19.  Tested numerous times to confirm.       

### Steps to Reproduce

1. Run M48 E

**Expected behavior:** Probe should only move when commanded to do so.

**Actual behavior:** Probe deploys/retracts unexpectedly and erratically.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Reywas62, 2019-09-25:
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/3653651/Configuration_adv.zip)
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3653652/Configuration.zip)

shitcreek, 2019-09-25:
Just tested M48 E with a BLTouch without any issue. Probed ten times and returned deviation amount. Did you homed before issuing M48? Did you have any issue with G29?

Reywas62, 2019-09-25:
The issue isn't specifically with M48. The problem occurs sporadically whenever the probe is deployed.  M48 E was just an easy way to activate the probe multiple times without having to do it manually.  As stated above, the issue does not occur with an earlier version of bugfix.

boelle, 2019-10-21:
what version of bugfix do you use? 1.1.x or 2.0.x ?
where do you download from? please give direct link to page

Reywas62, 2019-10-21:
bugfix-2.0.x downloaded from http://marlinfw.org/docs/basics/install.html

boelle, 2019-10-21:
what kind of probe?

boelle, 2019-10-21:
do you have 

//#define DEACTIVATE_SERVOS_AFTER_MOVE

or

#define DEACTIVATE_SERVOS_AFTER_MOVE

?

Reywas62, 2019-10-21:
Servo Deployed Z-probe, old-school.
#define DEACTIVATE_SERVOS_AFTER_MOVE

boelle, 2019-10-21:
then there are only 2 options left

either bad hardware or a fault in marlin, 

are the wires and connections good? crimped connections and not soldered? what voltage do the servo get? can it get enough juice?

Reywas62, 2019-10-21:
As I said in the original post, the hardware works fine with an earlier version of bugfix (pulled on 6/22/19).

boelle, 2019-10-21:
do you know the precise date it stopped working?

Reywas62, 2019-10-21:
If you mean what pull-date it stopped working then no.  It is still working with the 6/22/19 pull.  When I have the time I plan on trying to figure out the specific pull date that it stopped working.

boelle, 2019-10-22:
that would help the coders a great deal :-D

boelle, 2019-10-30:
figured out the pull date?

boelle, 2019-11-03:
closing this one until we have the pull date

we can reopen when we have the date  as that is important, without it we cant narrow it down

closed 2019-11-03 by boelle