MarlinFirmware/Marlin #15377: Incorrect PIN assignment in pins_RAMPS.h

julianwelbyeverard, 2019-09-26:
### Description

Incorrect PIN assignment in pins_RAMPS.h (both bugfix 1.1.x and bugfix 2.0.x branches)

When using a MKS_MINI_12864 display on a BOARD_MKS_GEN_L board that the assignment pin number in pint_RAMPS.h
#define KILL_PIN          41
should be
#define KILL_PIN          64

both my Display and Board are Original and not clones

### Steps to Reproduce

set the board to 
#define MOTHERBOARD BOARD_MKS_GEN_L
Display to
#define MKS_MINI_12864

Compile and upload

**Expected behaviour:** 
The display should show boot screen and information etc

**Actual behaviour:**
The does not show boot screen and information etc

#### Additional Information
Change the pin in pins_RAMPS as above #define KILL_PIN          64 and display now works
(note: I am not sure of the effect elsewhere of this change but in my tests, all seems ok)
(Hardware, MKS GEN L board, MKS MINI 12864, TMC2208_STANDALONE on XYZ and A4988 on E, Dual Z on E1 TMC2208_STANDALONE , BLTouch as per example setup)

thinkyhead, 2019-09-26:
- Do you think this will apply to all RAMPS 1.3 - 1.6 boards?
  Or, is this just limited to the MKS Gen-L?
- Does the MKS Gen-L route pin 64 to the spare button on all other types of displays that the generic RAMPS routes pin 41 to?

julianwelbyeverard, 2019-09-26:
That I don't know as I only have MKS Gen-L boards, I found this late last night and have not had the time dig through code in more depth, hopefully, someone might be able to check on another board,

GMagician, 2019-09-26:
I have a mega2560 with ramps and kill button on lcd is working (as I remember) so I think this doesn't apply to all ramp boards

thisiskeithb, 2019-09-27:
Using the latest `bugfix-2.0.x` commit (665e45e0bad4254352330884a846c21aad2a2ff0) and using the [Tevo Michelangelo config](https://github.com/MarlinFirmware/Marlin/tree/665e45e0bad4254352330884a846c21aad2a2ff0/config/examples/Tevo/Michelangelo) (`BOARD_MKS_GEN_L` & `MKS_MINI_12864`) & flashing it to my MKS GEN-L with MKS Mini 12864 LCD, the kill pin works fine and the bootscreen is shown/the LCD works fine as well.

thinkyhead, 2019-09-29:
@julianwelbyeverard — Is there any extra version information silkscreened onto your board? It sounds like you might have a unicorn.

julianwelbyeverard, 2019-09-29:
> 
> 
> @julianwelbyeverard — Is there any extra version information silkscreened onto your board? It sounds like you might have a unicorn.

[image]
[image]
[image]

Attached is both photos of the front and back of the boards.. hope this helps

boelle, 2019-10-21:
@julianwelbyeverard did you try what @thisiskeithb suggested?

and have you tried latest bugfix 2.0.x ?

closed 2019-10-24 by thinkyhead