MarlinFirmware/Marlin #15379: G28 Homing Issue when specifying axis

athulsnair, 2019-09-26:
### Description

When trying to home any two axis of the 3d printer, only first parameter is executed. 
For example, as per the documentation G28 [axis1] [axis2] should home both the [axis1] & [axis2] axis, but only homes [axis 1].
eg : G28 X Y ==> Homes X
       G28 Y X ==> Homes Y

one strange behaviour, 
        G28 X Y Z ==> homes X & Z

Adding 0 with argument works, 
eg: 
G28 X0 Y0 ==> Homes X & Y

### Steps to Reproduce

1. Send G28 X Y to printer

**Expected behavior:** [
printer should home both X & Y

**Actual behavior:** 
Printer homes only X axis not Y

#### Additional Information

Board used RAMPS 1.4, end stops are limit switches. Tested on two machines running at 1.1.6 & 1.1.9. One machine use limit switches as end stop, other use BLTouch as Z axis probe(Z probe on Z_MIN). Limit switch statues are checked on both machines before print.

Verified by another user from Marlin Facebook group,
see this post,
[Marlin Facebook Group](https://www.facebook.com/groups/1049718498464482/permalink/1892827747486882/)

sl1pkn07, 2019-09-26:
is same behavoir i have in here https://github.com/MarlinFirmware/Marlin/pull/15179#issuecomment-532415168

greetings

athulsnair, 2019-09-27:
In my case drivers are Polulu A4988, and DRV8825. Both boards use mix of these drivers

boelle, 2019-10-12:
@athulsnair still having issues?

boelle, 2019-10-21:
will close this one as it seems @athulsnair dont have an issue
i'm also not able to reproduce, homing individual axis works fine

we can always reopen if there is an issue

closed 2019-10-21 by boelle