MarlinFirmware/Marlin #15381: Writing G29 mesh offset gives incorrect results

Bernie-X, 2019-09-26:
### Description

I am getting incorrect mesh data after I write a mesh z offset to a particular grid point.

### Steps to Reproduce

1. Home, bilinear probe, and receive report

- G28 G29

2. Want to change mesh data for grid point I2 J0 from -1.521 to -1.519

- G29 W1 I2 J0 Z.02 

3. Get new report

- M420 V

**Expected behavior:**
I would expect I2 J0 to be -1.519

**Actual behavior:**

I2 J0 is -.876 (or thereabout)

#### Additional Information

Using bugfix-1.1.9

shitcreek, 2019-10-07:
You need to follow the template and provide your configuration files.
Marlin 1.x isn't worked on anymore, please try Marlin 2.0 and report if issue persists.

thinkyhead, 2019-10-09:
The code of `G29 W` in 1.1.9 looks okay to me. It's quite simple. I think you might be seeing a spurious glitch in the matrix.

Bernie-X, 2019-10-09:
Thanks all for your insight.  I figured out that I need to factor in my home offset in calculating the desired mesh offset.  Hence, Mesh<sub>offset</sub> = Home<sub>z offset</sub> + Mesh<sub>ij</sub>

closed 2019-10-09 by Bernie-X