MarlinFirmware/Marlin #15384: [BUG] 3DTouch sensor signal doesn't seem to work

Kurolox, 2019-09-26:
The 3DTouch sensor (a BLTouch clone) doesn't seem to react to relevant Gcodes, like M401 or M402, or any BLTouch option on the LCD menu.

The probe self tests itself properly when powering up, with the blue and red LEDs staying ON until powered off.

I've checked the cables multiple times and they're properly wired to the board (SKR v1.3)

I think it's worth mentioning that I've found this issue mentioned multiple times already, but I believe it's different enough to warrant it's own issue. https://github.com/MarlinFirmware/Marlin/issues/13890 has issues randomly while I'm unable to use the probe at all, and my issue looks quite similar to https://github.com/MarlinFirmware/Marlin/issues/14424, but since the author of said issue didn't really provide a lot of relevant information, I'm making a new issue.

It was mentioned in said issue to enable PINS_DEBUGGING to test, but I'm unable to do so due to errors compiling when enabling said setting (pins/pins.h:964:21: error: 'E1_MS3_PIN' was not declared in this scope)

https://github.com/MarlinFirmware/Marlin/issues/13861 is also similar to my issue, but I don't have FAST_PWM_FAN enabled, so it's probably unrelated.

### Steps to Reproduce

1. Connect a 3DTouch sensor to the servo pins on the SKR v1.3
2. Use the configuration files provided in the issue
3. Compile the firmware and upload it to the printer
4. Reset EEPROM settings (optional)

**Expected behavior:** The probe should work

**Actual behavior:** The probe doesn't react to anything

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files. [configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3659487/configuration.zip)
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thinkyhead, 2019-09-27:
I don't think we can help you here. It sounds like you got a bad probe, or possibly you need to swap the wires in the connector so they match your servo plug. Some of these BLTouch and clones are not wired exactly right.

Kurolox, 2019-09-27:
The probe works properly. It was installed to a RAMPS 1.4 previously, and I've attached the probe to the  RAMPS again to check. It's working, so it's not a hardware issue.

Regarding the connectors, I already said that I checked the cables and they're properly wired. I had to reverse the 5V and GND pins on the connector since they were inverted, but their current pinout matches the one at the board.

thinkyhead, 2019-09-27:
Given that you are the individual experiencing the issue, we will be counting on you to play around with the settings, code, timings, and so on to tell us what is causing the problem. I'm not sure what we can do without further data.

Kurolox, 2019-09-27:
Fair enough. I've been tinkering for a few days already, so this is what I've got:

- As previously stated, the probe works fine and the connection is working too
- I've tried increasing the BLTOUCH_DELAY from 200 to 1000 gradually, but there was no difference
- I've tried to enable manually the servo subsystem with NUM_SERVOS and SERVO_DELAY, no difference
- I've measured the voltage provided to the probe by the board (4.89v, so looks fine)
- I've plugged a servo (inverting the GND and 5v wires too) and tried the BLTouch commands. The servo seems to react to most of them, so there's some data transmission going on, and the board servo pinout seems to be working properly
- I've double checked that the servo pin is properly defined in pins.h (though it wasn't probably needed since the servo worked properly)
- I've enabled PINS_DEBUGGING (it seems to refuse to compile if you also have DUAL_Z_CARRIAGE enabled, I should probably report this as a standalone bug) and sent an M43 S command, both with and without Z_MIN_ENDSTOP_INVERTING and Z_MIN_PROBE_ENDSTOP_INVERTING toggled. Here are the outputs:

```
>>>M43 S
SENDING:M43 S
Servo probe test
. using index:  0, deploy angle: 10, stow angle:   90
. Probe Z_MIN_PIN: 57
. Z_MIN_ENDSTOP_INVERTING: false
. Check for BLTOUCH
. Deploy & stow 4 times
** Please trigger probe within 30 sec **
FAIL: No trigger detected
```

```
>>>M43 S
SENDING:M43 S
Servo probe test
. using index:  0, deploy angle: 10, stow angle:   90
. Probe Z_MIN_PIN: 57
. Z_MIN_ENDSTOP_INVERTING: true
. Check for BLTOUCH
. Deploy & stow 4 times
** Please trigger probe within 30 sec **
FAIL: No trigger detected
```

The probe doesn't react to the command at all. I've tried manually toggling the pin 200 with M43, but nothing was appreciated either.

That's all I've been able to check for now, but I'm somewhat confused about it. Everything seems to be fine on paper, from the configuration and the board (since the servo works) to the probe itself (since it works in a different board), but something is failing somewhere and I'm not being able to find what's the issue.

ebraiman, 2019-09-28:
This setup might help ... https://youtu.be/5cSzFCv7K4Q

Kurolox, 2019-09-29:
> This setup might help ... https://youtu.be/5cSzFCv7K4Q

That's the same setup I have. 

Either there's a bug in Marlin on how the probe is emitting the signal to the probe, or the board has a manufacturing issue where the signal is being deformed and the probe isn't recognizing it.

I don't really have the tools at hand to check for the second option, though.

boelle, 2019-10-21:
if it was me i would test the input on the board (endstop input pins) with a microswitch to confirm if marlin is the fault

if i remember correct these probe work with the servo connector to give the probe commands
and then the probe send signal to the endstop, this signal can be either high or low

use the M119 command to check the status of the endstop

Kurolox, 2019-10-21:
The servo signal works, as I said before I tried plugging a servo and the servo was showing movement when issuing signals to the 3dtouch.

At this point I'm still unable to figure out what is the issue. I've spoken with geetech support and all I was able to figure out is that the probe signal operates at 5V rather than the board 3.3V, and disassemblying the probe shows a cutable pad to change the signal voltage (pretty much like in the BLTouch, but undocumented). Cutting this pad, however, didn't solve the issue.

I've also talked with bigtreetech support, and while they've been way more helpful and responsive (even sent me another board just to rule out if this one was defective), they couldn't really do anything and it doesn't seem like an issue on their end.

At this point I just bought a genuine BLTouch 3.0 probe and I'm waiting for it to arrive. If that probe works, then I guess the issue is that there's some essential difference between the BLTouch probe and the 3DTouch that makes it unable to work in this specific board, but again, the geetech support wasn't really helpful in providing any relevant information about this.

I'll close this issue when the probe arrives and I can confirm if this was the cause.

Kurolox, 2019-10-28:
Sorry for being late. The BLTouch 3.0 works, so it's something in the 3dtouch design that makes it incompatible with the SKR v1.3 board specifically.

I'm not sure if anything can be done in the firmware side since I'm still unable to find out what's exactly the issue, so I'll close this for now.

closed 2019-10-28 by Kurolox

opiswahn, 2019-11-04:
@Kurolox  I have the same Problem with a 3DTouch(bought is 2 years ago i think...). It always worked fine on Marlin 1.1.9 with a MKS Gen-L board, now i switched to a SKR 1.3 and Marlin 2.0, but it isn't even self-testing anymore, it behaves just like you described in the description.
Is there any solution to this other than buying a new one?

Kurolox, 2019-11-04:
Since I wasn't able to determine where the issue lies, I can't really say if there's anything that can be done to fix it besides replacing it with a genuine one.

Geetech hasn't been really helpful when adressing this issue, but you can try contacting them about it.

opiswahn, 2019-11-05:
@Kurolox i now managed to get it working by disabling FAST_PWM_FAN... this is all pretty strange, but is seems to have to do something with the sensor itself. A original BLTouch (v3) from a friend worked even with FAST_PWM_FAN turned on.
So the Combination of some cloned version and FAST_PWM_FAN is the issue here, so i don't thin the firmware is to blame but a bad implementation of the 3DTouch firmware ot it's hardware.