MarlinFirmware/Marlin #15385: [BUG] Menu not available during printing

hapklaar, 2019-09-26:
### Description

When I press the knob on the RepRapDiscount Smart Controller during printing, I get a blank screen which can't be exited. It will automatically return to status screen after a while. Accessing the menu works fine when there is no print in progress

### Steps to Reproduce

1. Start print through Octoprint or from SD
2. Press menu knob on panel
3. behold blank LCD screen

**Expected behavior:** Show menu

**Actual behavior:** Blank LCD screen

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3659826/Configuration.zip)

- Anycubic Kossel Plus
- SKR 1.3 Board
- X/Y/Z/E0 = TMC2130 /w software SPI
- latest bugfix 2.0 code used

hapklaar, 2019-09-26:
Seems I'm hit by the same issue as in #15045

Will try the fix mentioned there by setting BLOCK_BUFFER_SIZE to 64 and report back.

hapklaar, 2019-09-26:
Yup, that fixed it. Might be something that can be tied to 32 bit board types.

tpruvot, 2019-09-28:
was 16 ? did you try with 32 ?

hapklaar, 2019-09-28:
No, someone did try that however but wasn't successful (https://github.com/MarlinFirmware/Marlin/issues/12461#issuecomment-525027829)

I went to 64 immediately as I saw no downside to it.

closed 2019-10-03 by shitcreek

hapklaar, 2019-10-03:
Why is this closed, it's obviously not the same issue as in #15414 as this fix didn't work there

shitcreek, 2019-10-03:
You said changing the buffer to 64 fixed your issue. The other issue is still open.