MarlinFirmware/Marlin #15386: Marlin 2.0 can't complain when enabling neopixel

brasco901, 2019-09-27:
whenever I try and compile  by defining #define NEOPIXEL_LED  I get major errors on the compile. 

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/master.zip; framework: arduino; bo
ard: nxp_lpc1768)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x 0.0.2 > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.80KB RAM, 464KB Flash

DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES: framework-arduino-lpc176x 0.1.3, toolchain-gccarmnoneeabi 1.80201.181220 (8.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.0
|-- <U8glib-HAL> 0.4
|-- <TMCStepper> 0.5.0
|-- <Adafruit NeoPixel> 1.2.4
|-- <SailfishLCD>
 
Unable to find destination disk (Autodetect Error)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/sectio
n_env_upload.html) or copy the firmware (.pio/build/LPC1768/firmware.bin) manually to the appropriate disk
 
Compiling .pio/build/LPC1768/src/src/HAL/HAL_LPC1768/DebugMonitor_LPC1768.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/HAL_LPC1768/HAL.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/HAL_LPC1768/HAL_spi.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/HAL_LPC1768/HAL_timers.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/HAL_LPC1768/MarlinSerial.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/HAL_LPC1768/fast_pwm.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/HAL_LPC1768/include/digipot_mcp4451_I2C_routines.c.o
Compiling .pio/build/LPC1768/src/src/HAL/HAL_LPC1768/main.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/HAL_LPC1768/persistent_store_flash.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/HAL_LPC1768/persistent_store_sdcard.cpp.o
In file included from Marlin/src/HAL/HAL_LPC1768/../../core/../inc/MarlinConfig.h:38,
                 from Marlin/src/HAL/HAL_LPC1768/../../core/serial.h:24,
                 from Marlin/src/HAL/HAL_LPC1768/DebugMonitor_LPC1768.cpp:26:
Marlin/src/HAL/HAL_LPC1768/../../core/../inc/SanityCheck.h:1827:1: error: user-defined literal in preprocessor expression
   #if !(PIN_EXISTS(NEOPIXEL) && NEOPIXEL_PIXELS > 0)
 ^ ~~~~~~~~~~
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:38,
                 from Marlin/src/HAL/HAL_LPC1768/HAL.cpp:25:
Marlin/src/HAL/HAL_LPC1768/../../inc/SanityCheck.h:1827:1: error: user-defined literal in preprocessor expression
   #if !(PIN_EXISTS(NEOPIXEL) && NEOPIXEL_PIXELS > 0)
 ^ ~~~~~~~~~~
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:38,
                 from Marlin/src/HAL/HAL_LPC1768/HAL_spi.cpp:51:
Marlin/src/HAL/HAL_LPC1768/../../inc/SanityCheck.h:1827:1: error: user-defined literal in preprocessor expression
 #if !(PIN_EXISTS(NEOPIXEL) && NEOPIXEL_PIXELS > 0)
 ^ ~~~~~~~~~~
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:38,

                 from Marlin/src/HAL/HAL_LPC1768/HAL_timers.cpp:31:
Marlin/src/HAL/HAL_LPC1768/../../inc/SanityCheck.h:1827:1: error: user-defined literal in preprocessor expression
   #if !(PIN_EXISTS(NEOPIXEL) && NEOPIXEL_PIXELS > 0)
 ^ ~~~~~~~~~~
In file included from Marlin/src/HAL/HAL_LPC1768/../../sd/../inc/MarlinConfig.h:38,
                 from Marlin/src/HAL/HAL_LPC1768/../../sd/cardreader.h:24,
                 from Marlin/src/HAL/HAL_LPC1768/main.cpp:38:
Marlin/src/HAL/HAL_LPC1768/../../sd/../inc/SanityCheck.h:1827:1: error: user-defined literal in preprocessor expression
   #if !(PIN_EXISTS(NEOPIXEL) && NEOPIXEL_PIXELS > 0)
 ^ ~~~~~~~~~~
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:38,
                 from Marlin/src/HAL/HAL_LPC1768/persistent_store_flash.cpp:44:
Marlin/src/HAL/HAL_LPC1768/../../inc/SanityCheck.h:1827:1: error: user-defined literal in preprocessor expression
   #if !(PIN_EXISTS(NEOPIXEL) && NEOPIXEL_PIXELS > 0)
 ^ ~~~~~~~~~~
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:38,
                 from Marlin/src/HAL/HAL_LPC1768/persistent_store_sdcard.cpp:29:
Marlin/src/HAL/HAL_LPC1768/../../inc/SanityCheck.h:1827:1: error: user-defined literal in preprocessor expression
   #if !(PIN_EXISTS(NEOPIXEL) && NEOPIXEL_PIXELS > 0)
 ^ ~~~~~~~~~~
Compiling .pio/build/LPC1768/src/src/HAL/HAL_LPC1768/u8g/u8g_com_HAL_LPC1768_hw_spi.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/HAL_LPC1768/u8g/u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp.o
*** [.pio/build/LPC1768/src/src/HAL/HAL_LPC1768/DebugMonitor_LPC1768.cpp.o] Error 1
*** [.pio/build/LPC1768/src/src/HAL/HAL_LPC1768/HAL.cpp.o] Error 1
*** [.pio/build/LPC1768/src/src/HAL/HAL_LPC1768/HAL_timers.cpp.o] Error 1
*** [.pio/build/LPC1768/src/src/HAL/HAL_LPC1768/HAL_spi.cpp.o] Error 1
*** [.pio/build/LPC1768/src/src/HAL/HAL_LPC1768/main.cpp.o] Error 1
*** [.pio/build/LPC1768/src/src/HAL/HAL_LPC1768/persistent_store_flash.cpp.o] Error 1
*** [.pio/build/LPC1768/src/src/HAL/HAL_LPC1768/persistent_store_sdcard.cpp.o] Error 1
========================== [FAILED] Took 5.66 seconds ==========================
 
Environment               Status    Duration
------------------------  --------  ------------
megaatmega2560            IGNORED
megaatmega1280            IGNORED
at90usb1286_cdc           IGNORED
at90usb1286_dfu           IGNORED
DUE                       IGNORED
DUE_USB                   IGNORED
DUE_debug                 IGNORED
LPC1768                   FAILED    00:00:05.664

brasco901, 2019-09-27:
[brasco901configs.zip](https://github.com/MarlinFirmware/Marlin/files/3660449/brasco901configs.zip)
Attached my configs

brasco901, 2019-09-27:
i fixed it.  my issue was that my neopixel pin was defined wrong in config.h

it should be 
#define NEOPIXEL_PIN SERVO0_PIN

p3p, 2019-09-27:
The issue was using `2_00` instead of `P2_00`, but using `SERVO0_PIN` will pix it in a round about way.

closed 2019-09-27 by p3p