MarlinFirmware/Marlin #15388: [BUG] SKR 1.3 SPI 2130 Dual Z Z2 E1 Temp Conflict

motion-au, 2019-09-27:
## Description

SKR 1.3 Board with Marlin 2.0 Stock Settings, correctly changed configuration to enable Dual Z stepper via the E1 driver to enable Z2.

### Steps to Reproduce

Set E1 as the driver for Z2

**Expected behavior:** Using pronterface to move the Z axis, with Z and E1 plugged into the Z axis motors, both should rotate in tandem.

**Actual behavior:** Only Z rotates, E1 was unresponsive.

#### Additional Information

Currently assembling the frame (testing individual components), no thermistors are plugged in to the board so Extruder temp was -14 deg C. I had to lower it to -15 deg C in the config and E0 and E1 became functional. This is equilivant to disabling the prevent_cold_extrude function, if I understand correctly. This setting appeared to interfere with E1 despite the program settings assigning it as Z2. The Z axis is now functioning with Z and E1 outputs - again, it wasnt until I changed the extrude minimum temperature despite allocating E1 as Z2.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3662007/Configuration.zip)

Sorry about any issues relating to the format of this, first post.

JJHMARR2020, 2019-09-27:
Find pins_Bigtree_SKR_V1.3.h under lpc1768 under pins in the src folder
Change:

#define E1_STEP_PIN        P0_01
#define E1_DIR_PIN         P0_00
#define E1_ENABLE_PIN      P0_10
#ifndef E1_CS_PIN
  #define E1_CS_PIN        P1_01
#endif

To:
#define Z2_STEP_PIN        P0_01
#define Z2_DIR_PIN         P0_00
#define Z2_ENABLE_PIN      P0_10
#ifndef Z2_CS_PIN
  #define Z2_CS_PIN        P1_01
#endif

I had a similar issue on the SKR Pro V1.1 Changing the name of the controller should solve the issue.

boelle, 2019-10-12:
@motion-au still having issues?

motion-au, 2019-10-23:
Fixed. Thank you

closed 2019-10-23 by motion-au

Talha909, 2019-12-04:
> Fixed. Thank you

how you fixed this issue.