MarlinFirmware/Marlin #15389: [BUG] Compile error on #define PHOTO_GCODE

martinglass, 2019-09-27:
### Description

Compile error if #define PHOTO_GCODE is uncommented out

### Steps to Reproduce

1. uncomment this line in Configuration_adv.h
#define PHOTO_GCODE

2. Verify the code in Arduino

**Expected behavior:** [What you expect to happen]
No errors

**Actual behavior:** [What actually happens]
The compiler returns the following errors:
Arduino: 1.8.10 (Mac OS X), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch/src/HAL/HAL_AVR/../../inc/MarlinConfig.h:38:0,
                 from sketch/src/HAL/HAL_AVR/MarlinSerial.cpp:39:
sketch/src/HAL/HAL_AVR/../../inc/SanityCheck.h:2364:1: error: operator "defined" requires an identifier
   #if (PIN_EXISTS(CHDK) + PIN_EXISTS(PHOTOGRAPH_PIN) + defined(PHOTO_SWITCH_POSITION)) > 1
 ^ ~~~~
In file included from sketch/src/HAL/HAL_AVR/../../inc/MarlinConfig.h:38:0,
                 from sketch/src/HAL/HAL_AVR/HAL.cpp:25:
sketch/src/HAL/HAL_AVR/../../inc/SanityCheck.h:2364:1: error: operator "defined" requires an identifier
   #if (PIN_EXISTS(CHDK) + PIN_EXISTS(PHOTOGRAPH_PIN) + defined(PHOTO_SWITCH_POSITION)) > 1
 ^ ~~~~
sketch/src/HAL/HAL_AVR/../../inc/SanityCheck.h:2364:1: error: user-defined literal in preprocessor expression
sketch/src/HAL/HAL_AVR/../../inc/SanityCheck.h:2364:86: error: missing '(' in expression
   #if (PIN_EXISTS(CHDK) + PIN_EXISTS(PHOTOGRAPH_PIN) + defined(PHOTO_SWITCH_POSITION)) > 1
                                                                                      ^
In file included from sketch/src/inc/MarlinConfig.h:38:0,
                 from sketch/src/Marlin.h:24,
                 from sketch/src/Marlin.cpp:31:
sketch/src/inc/SanityCheck.h:2364:1: error: operator "defined" requires an identifier
   #if (PIN_EXISTS(CHDK) + PIN_EXISTS(PHOTOGRAPH_PIN) + defined(PHOTO_SWITCH_POSITION)) > 1
 ^ ~~~~
sketch/src/inc/SanityCheck.h:2364:1: error: user-defined literal in preprocessor expression
sketch/src/inc/SanityCheck.h:2364:86: error: missing '(' in expression
   #if (PIN_EXISTS(CHDK) + PIN_EXISTS(PHOTOGRAPH_PIN) + defined(PHOTO_SWITCH_POSITION)) > 1
                                                                                      ^
sketch/src/HAL/HAL_AVR/../../inc/SanityCheck.h:2364:1: error: user-defined literal in preprocessor expression
sketch/src/HAL/HAL_AVR/../../inc/SanityCheck.h:2364:86: error: missing '(' in expression
   #if (PIN_EXISTS(CHDK) + PIN_EXISTS(PHOTOGRAPH_PIN) + defined(PHOTO_SWITCH_POSITION)) > 1
                                                                                      ^
In file included from sketch/src/HAL/HAL_AVR/../../inc/MarlinConfig.h:38:0,
                 from sketch/src/HAL/HAL_AVR/HAL_SPI.cpp:34:
sketch/src/HAL/HAL_AVR/../../inc/SanityCheck.h:2364:1: error: operator "defined" requires an identifier
   #if (PIN_EXISTS(CHDK) + PIN_EXISTS(PHOTOGRAPH_PIN) + defined(PHOTO_SWITCH_POSITION)) > 1
 ^ ~~~~
sketch/src/HAL/HAL_AVR/../../inc/SanityCheck.h:2364:1: error: user-defined literal in preprocessor expression
sketch/src/HAL/HAL_AVR/../../inc/SanityCheck.h:2364:86: error: missing '(' in expression
   #if (PIN_EXISTS(CHDK) + PIN_EXISTS(PHOTOGRAPH_PIN) + defined(PHOTO_SWITCH_POSITION)) > 1
                                                                                      ^
In file included from sketch/src/HAL/HAL_AVR/../../inc/MarlinConfig.h:38:0,
                 from sketch/src/HAL/HAL_AVR/Servo.cpp:56:
sketch/src/HAL/HAL_AVR/../../inc/SanityCheck.h:2364:1: error: operator "defined" requires an identifier
   #if (PIN_EXISTS(CHDK) + PIN_EXISTS(PHOTOGRAPH_PIN) + defined(PHOTO_SWITCH_POSITION)) > 1
 ^ ~~~~
sketch/src/HAL/HAL_AVR/../../inc/SanityCheck.h:2364:1: error: user-defined literal in preprocessor expression
sketch/src/HAL/HAL_AVR/../../inc/SanityCheck.h:2364:86: error: missing '(' in expression
   #if (PIN_EXISTS(CHDK) + PIN_EXISTS(PHOTOGRAPH_PIN) + defined(PHOTO_SWITCH_POSITION)) > 1
                                                                                      ^
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

#### Additional Information

None

InsanityAutomation, 2019-09-28:
PR to fix this is open

martinglass, 2019-09-28:
All fixed - many thanks guys

closed 2019-09-28 by martinglass