MarlinFirmware/Marlin #15391: [BUG]Reprap Discount Full Graphics Smart Controller Display is Garbled

roland1970, 2019-09-27:
Hello

I have downloaded 2.0 bugfix ,my display is an Reprap Discount Full Graphics Smart Controller Display .
Problem with the display is that is garbled after flashing the firmware.
That problem did I also have with version 1.9 and that is solved by the following instruction.
[code] 
 U8GLIB_ST7920_128X64_RRD u8g(0); // Number of stripes can be adjusted in ultralcd_st7920_u8glib_rrd.h with PAGE_HEIGHT              

 U8GLIB_ST7920_128X64_1X u8g(23,17,16); // Number of stripes can be adjusted in ultralcd_st7920_u8glib_rrd.h with PAGE_HEIGHT       [/code] 
The codes located in ultralcd_impl_DOGM.h
The first line did I disabled en copy and adjusted in the second line .
So did I fix that the display works .
Did anyone a suggestion how to fix my problem in 2.0 ?

Printer is a Hypercube Evo with Arduino en Ramps board.

ALEX-45, 2019-10-05:
Yes, there is a problem. But there is no file "ultralcd_impl_DOGM.h", and I don’t know how to fix it myself.

ALEX-45, 2019-10-05:
recent bug, but a very important issue. I observe in 2.0 and 1.1.x

Coollision, 2019-10-16:
just spent some hours on it as wel, found a fix for marlin 2.0, in the file src/lcd/dogm/ultralcd_DOGM.h on line 54 (you'l see somewhere above that line a comment saying RepRap Discount Full Graphics Smart Controller) change the line from #define U8G_CLASS U8GLIB_ST7920_128X64_4X
to #define U8G_CLASS U8GLIB_ST7920_128X64_1X
that worked for me

ALEX-45, 2019-10-17:
I dealt with a bug adding the following lines in the configuration.h

//my add исправление дисплея https://3dtoday.ru/questions/artefakty-na-lcd-12864
#define ST7920_DELAY_1 DELAY_NS(0)
#define ST7920_DELAY_2 DELAY_NS(188) 
#define ST7920_DELAY_3 DELAY_NS(0)

It was like this:
[image]

Fixed.

Coollision, 2019-10-17:
Indeed forgot to mention that, did that as well.

boelle, 2019-10-22:
@Coollision maybe submit a PR so that everyone can benefit?

InsanityAutomation, 2019-10-22:
@boelle that doesnt make sense to do. The defaults work for most. When you get something that it doesnt it needs adjusting.

boelle, 2019-10-22:
just a small brainfart from my side

so this is an "edge" case? if so should we just close this one or is there a better action to take?

ALEX-45, 2019-10-22:
in theory, you need to make a choice from macro definitions for different displays (if they are different) so that the firmware works out of the box. In the old versions everything worked and so, and in the new ones the timings were apparently cut. Thus, there is a bug, and at the moment it is fixed only with a crutch, since most ordinary users can only uncomment or comment on definitions and change numerical values in the configuration file.

Coollision, 2019-10-22:
I think indeed the timing are there to adjust on a per user basis, but perhaps someone with more experience could look into my adjustment of the ultralcd_DOGM.h file?

boelle, 2019-10-24:
@shitcreek time to have a quick look ?

viebvo, 2019-10-28:
Just wanted to add that i had the same problem with current 2.0 and the fix from @Coollision worked for me.

Coollision, 2019-10-29:
@viebvo only the fix of the src/lcd/dogm/ultralcd_DOGM.h or did you adjust the timings as well?

viebvo, 2019-10-29:
@Coollision - just changed from "#define U8G_CLASS U8GLIB_ST7920_128X64_RRD" to "#define U8G_CLASS U8GLIB_ST7920_128X64_1X"
The file ultralcd_impl_DOGM.h is missing in the current 2.0 sourcetree.

boelle, 2019-11-03:
@InsanityAutomation would it be fair to close this one as it could fall under the per user case?

closed 2019-11-07 by roland1970

MechyJoel, 2020-01-24:
Roland1970 suggested fix worked for me as well. Had to search ultralcd in the marlin folder to find the appropriate .h file to edit.

Whitke, 2020-01-26:
I know this is closed, but just wanted to thank the group for the info! @Coollision your formula worked to clean up my LCD screen. 
Thank you!

zalexzperez, 2020-03-17:
> I dealt with a bug adding the following lines in the configuration.h
> 
> //my add исправление дисплея https://3dtoday.ru/questions/artefakty-na-lcd-12864
> #define ST7920_DELAY_1 DELAY_NS(0)
> #define ST7920_DELAY_2 DELAY_NS(188)
> #define ST7920_DELAY_3 DELAY_NS(0)
> 
> It was like this:
> [image]
> 
> Fixed.

Fixed my Ender3 on the latest bugfix 2.0.x nighlty compilation. Thank you!

ALEX-45, 2020-03-17:
> Fixed my Ender3 on the latest bugfix 2.0.x nighlty compilation. Thank you!

not at all