MarlinFirmware/Marlin #15392: [FR] Four Z Axis Motorized ABL, G34

gmedlicott, 2019-09-27:
Feature request for four Z axis motorized ABL support to G34 command (G34 currently supports three motorized Z axis, #11302).

Used for moving gantry printers, where the XY plane is somewhat flexible, and four points are required to define the plane.  Eg. Voron 2, ExoCube, DIY etc.  

RepRapFirmware has this feature.

InsanityAutomation, 2019-09-29:
Ive thought about this one in the past too. Limit really comes down to not many boards with enough drivers to do everything I want! lol I think this one will get done, but likely not before 2.0. Maybe a 2.1 milestone tag?

alieneila, 2020-03-12:
Now that Quad Z is in, I'd like to bump this =D

InsanityAutomation, 2020-03-12:
See open PR :)

closed 2020-06-30 by github-actions[bot]