MarlinFirmware/Marlin #15393: [BUG] NEOPIXEL_BKGD_LED_INDEX > 0 prevents compiling

jph290, 2019-09-27:
### Description

When NEOPIXEL_BKGD_LED_INDEX > 0, the code fails to compile.

### Steps to Reproduce

1. #define NEOPIXEL_BKGD_LED_INDEX 1

**Expected behavior:** Compiles

**Actual behavior:** Marlin\src\feature/leds/neopixel.h:83:22: error: 'static void Marlin_NeoPixel::set_pixel_color(uint16_t, uint32_t)' cannot be overloaded with 'static void Marlin_NeoPixel::set_pixel_color(uint16_t, uint32_t)'

#### Additional Information
Commenting this out @ line 72-74 allows compiling.  Looks like conditional function was attempted but not done.

#ifdef NEOPIXEL_BKGD_LED_INDEX
    static void set_pixel_color(const uint16_t n, const uint32_t c);
#endif

brianredbeard, 2019-09-28:
Possibly related to #15397 as submitted by @alexsomesan

brianredbeard, 2019-09-28:
cc @extesy

alexsomesan, 2019-09-28:
@brianredbeard I got the build working by bumping the Adafruit NeoPixel version to 1.2.5 in `platformio.ini` line 31 (up at the top in the `[common]` section). 

Nice to hear from you again, old man!

Evg33, 2019-09-28:
Index led working on SKR 1.3 only with commenting 3 lines.

Christophegriffo, 2019-09-28:
Cela fait un moment qu'il y a se problème. 
il vous faut commenter 
static void set_pixel_color (const uint16_t n, const uint32_t c);

//static void set_pixel_color (const uint16_t n, const uint32_t c);

Pour contourner le problème . C'est la solution qui me était rapporté

jph290, 2019-09-28:
I am running a bigtreetech SKR 1.3 BTW.  Just in case you guys needed to know.  Code didn't look like it cared so I didn't mention it.

Evg33, 2019-09-29:
SKR1.3
default_envs = LPC1768

Config:
```
  #define NEOPIXEL_BKGD_LED_INDEX  2               // Index of the LED to use
  #define NEOPIXEL_BKGD_COLOR { 0, 0, 255, 0 } // R, G, B, W
```

Error:
```
In file included from Marlin\src\feature/leds/leds.h:33,
                 from Marlin\src\Marlin.cpp:85:
Marlin\src\feature/leds/neopixel.h:83:22: error: 'static void Marlin_NeoPixel::set_pixel_color(uint16_t, uint32_t)' cannot be overloaded with 'static void Marlin_NeoPixel::set_pixel_color(uint16_t, uint32_t)'
   static inline void set_pixel_color(const uint16_t n, const uint32_t c) {
                      ^~~~~~~~~~~~~~~
Marlin\src\feature/leds/neopixel.h:73:17: note: previous declaration 'static void Marlin_NeoPixel::set_pixel_color(uint16_t, uint32_t)'
     static void set_pixel_color(const uint16_t n, const uint32_t c);
                 ^~~~~~~~~~~~~~~
```

If commenting line 73 in Marlin\src\feature\leds\neopixel.h 
```
  #ifdef NEOPIXEL_BKGD_LED_INDEX
//    static void set_pixel_color(const uint16_t n, const uint32_t c);
  #endif

```
Compilation success and BKGD_LED working.

thinkyhead, 2019-09-29:
'Twas broken by e5aa453293. Is now patched by 24706aedbd.

closed 2019-09-29 by thinkyhead