MarlinFirmware/Marlin #15399: [BUG] PINS_DEBUGGING conflicts with DUAL_STEPPER_DRIVERS

Kurolox, 2019-09-28:
### Description

It seems that if you try to enable PINS_DEBUGGING and Z_DUAL_STEPPER_DRIVERS (I haven't checked for X and Y, or triple Z) at the same time, the firmware won't compile due to the following errors:

```
In file included from Marlin/src/gcode/config/M43.cpp:29:0:
Marlin/src/gcode/config/../../inc/../pins/pins.h:964:21: error: 'E1_MS1_PIN' was not declared in this scope
 #define __EPIN(p,q) E##p##_##q##_PIN

Marlin/src/gcode/config/../../inc/../pins/pins.h:964:21: error: 'E1_MS2_PIN' was not declared in this scope
 #define __EPIN(p,q) E##p##_##q##_PIN

Marlin/src/gcode/config/../../inc/../pins/pins.h:964:21: error: 'E1_MS3_PIN' was not declared in this scope
 #define __EPIN(p,q) E##p##_##q##_PIN
```

It seems like the firmware is trying to access to the extruder E1 pins, which should've been redefined as Z1, and since they aren't defined it's refusing to compile.
- 

### Steps to Reproduce

1. Clone the repository and checkout the 2.0.0 branch (at commit https://github.com/MarlinFirmware/Marlin/commit/a6b0223eead964237e7920c35bb3c7757b722139 when writing this)
2. on `adv_configuration.h`, uncomment `//#define Z_DUAL_STEPPER_DRIVERS` and `//#define PINS_DEBUGGING`
3. Attempt to compile the firmware
**Expected behavior:** The firmware should compile

**Actual behavior:** The firmware fails to compile with the errors mentioned previously

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

InsanityAutomation, 2019-09-28:
I can confirm this, I ran into it the other day and didnt have time to chase further.

sjasonsmith, 2019-10-05:
As a workaround, you can define the following somewhere. I can't promise it won't cause problems somewhere, but I assume if you aren't using SPI drivers, these will never be used.

```
#define Z2_CS_PIN 0
#define Z2_MS1_PIN 0
#define Z2_MS2_PIN 0
#define Z2_MS3_PIN 0
```

If you have three Z steppers, do the same for Z3 as well.

otherguy, 2019-10-28:
@sjasonsmith thanks! I just ran into this as well.

The thing is, I _do_ have SPI drivers (TMC5160). What's the correct workaround here? Will it work with `PINS_DEBUGGING` disabled? It _compiles_, but will Z2 (E1) actually work?

boelle, 2019-11-09:
@sjasonsmith @otherguy any idea on how we could solve this one?

otherguy, 2019-11-09:
@boelle I wish I knew what the problem is.

I did not go for the workaround mentioned by @sjasonsmith because, as I expected, my SPI drivers didn’t work.

For now I have `Z_DUAL_STEPPER_DRIVERS` enabled and `PINS_DEBUGGING` disabled and this works.

sjasonsmith, 2019-11-09:
I think somebody fixed this. It seems to build fine for me.

@Kurolox or @otherguy, can one of you try it out with the latest commit from bugfix-2.0.x?

otherguy, 2019-11-11:
@sjasonsmith can confirm it works! Thank you -- can be closed.

boelle, 2019-11-12:
will close

add @ in front of my username if we should reopen

closed 2019-11-12 by boelle