MarlinFirmware/Marlin #15412: 3Dtouch Randomly deploys during grid leveling process and homing

Raymo99, 2019-09-28:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3665936/Marlin.zip)

### Description
3Dtouch clone probe randomly deploys during bed homing.  

### Steps to Reproduce
Using Marlin bugfix 1.1.x
Creality Ender 5 with standard board 
3Dtouch probe

**Expected behavior:**
I expect the probe should only deploy once per test point in the 3x3 grid

**Actual behavior:**
The probe randomly deploys then pops back up, 
It doesn't affect the leveling process, but seems like a bug.

#### Additional Information

Video of the behavior below and my config.h and adv.h are attached

https://photos.google.com/photo/AF1QipO5HW8eOeUfogYo-r5g-kSWnpskVKOJgF-tiHSY

happy to do any further testing as needed.

Ray

Raymo99, 2019-10-03:
added better video link

shitcreek, 2019-10-07:
Marlin 1.x is no longer worked on. Please try Marlin 2.x and report back if issue persist. Also, please be certain it is not a faulty device and not a configuration issue.

Quick glance of your configuration.h shows that the 'BLTOUCH_DELAY' is set to 75. Try increasing it to 300 or 500. Normally it does two

Raymo99, 2019-10-08:
sorry for the reply replying, it took me a while to figure out some issues while flashing 2.0, but its on there now.

Sadly the false deploy issue is still present.

The delay is Bltouch 500 as pre-defined in adv.h

What delayed me was not knowing that Define servo0_pin was implemented in  the pins creality file, but I got there eventually, you see I'm using pin 29 instead of pin 27...

shitcreek, 2019-10-08:
That's good. Are you all set now? If so, please close the thread. Thanks.

Raymo99, 2019-10-08:
Sorry, I may not have been clear enough... the problem is still present.
I still get false deploys of the probe.
Increased the delay to 1000, but the problem remains.

shitcreek, 2019-10-08:
What about increasing `DELAY_BEFORE_PROBING`?

Raymo99, 2019-10-08:
Good idea, but it still happens.
I used the default 200.

Raymo99, 2019-10-09:
I'm going to put an oscilloscope on the control line for the Bltouch and see what is actually happening.
I hope it reveal something, more to follow...

Aurawin, 2019-10-22:
As I mentioned in a prior thread - I have this problem too.  I took my existing wires from the V3 blt and replaced my BLT with a clone.  (So it's not a wiring problem).

I have worked around this issue by increasing the acceleration and feedrate of my z-axis.  I use the fast mode.  I have to say my probe repeatability is much more reliable with the clone.

It was worth the trouble for the workaround.  I probe z-dual-axis and 6 point bi-linear on each print with single shot (extra probe disabled).  Now my startup times are minuscule.

My touch pin still owers between probes when it should be raised as the probe moves from point to point.

Raymo99, 2019-10-22:
Hi Bo,
Sorry, I didn't get around to putting a scope in it yet. But since you ask tben I will and provide results. 

It will be a few days be6i can get to do it.

Cheers 
Ray

boelle, 2019-10-30:
@Raymo99 had time to play with the scope?

Aurawin, 2019-10-31:
> @Raymo99 had time to play with the scope?

I recently disabled my BLT FAST probing.  And since doing that my probe is 100% keeping the pin up during all probing moves.

boelle, 2019-10-31:
@Raymo99 maybe try what @Aurawin did and see if that makes it tick :-)

Raymo99, 2019-10-31:
Yes I will.
Sorry but I've been tied up with another project and have not revisited this issue. You know how it goes, new projects are so much more appealing than older ones.
I will see if disabling BLT Fast has any effect, otherwise I'll make scope recordings of the control line on video.

Raymo99, 2019-10-31:
It turns out I'm not using define BLTOUCH_HS_MODE or aka Fast.
so next step is to probe with a scope.

Raymo99, 2019-10-31:
Ok folks, scope testing is completed.
There are no command line pulses which match the erratic drop's.
Conclusion is a bad clone BLT.

Case closed, not a bug in Marlin.

Thanks for the support.