MarlinFirmware/Marlin #15413: [BUG] (linking issue for hal\hal_stm32\timers.h with stimer_t

happy77fr, 2019-09-28:
Hello, 

I am trying to compile marlin for a STM32F7 based board and I am facing the following (copied error message below)
As I am still trying to understand how everything is linked.
Can you help?
Thank you

Arduino: 1.8.10 (Windows 10), Board: "3D printer boards, RemRam v1, STM32CubeProgrammer (SWD), Enabled (no generic 'Serial'), CDC (no generic 'Serial'), Low/Full Speed, Smallest (-Os default), Newlib Nano (default)"

WARNING: library arduino_645681 claims to run on avr, sam architecture(s) and may be incompatible with your current board which runs on stm32 architecture(s).
WARNING: library TMCStepper claims to run on avr architecture(s) and may be incompatible with your current board which runs on stm32 architecture(s).
In file included from c:\users\USER\appdata\local\temp\arduino_build_947453\sketch\src\hal\hal_stm32\hal.h:99,
                 from c:\users\USER\appdata\local\temp\arduino_build_947453\sketch\src\hal\hal.h:26,
                 from sketch\src\inc/MarlinConfig.h:30,
                 from sketch\src\Marlin.h:24,
                 from sketch\src\Marlin.cpp:31:
c:\users\USER\appdata\local\temp\arduino_build_947453\sketch\src\hal\hal_stm32\timers.h:122:26: error: variable or field 'Step_Handler' declared void
 extern void Step_Handler(stimer_t *htim);
                          ^~~~~~~~
c:\users\USER\appdata\local\temp\arduino_build_947453\sketch\src\hal\hal_stm32\timers.h:122:26: error: 'stimer_t' was not declared in this scope
c:\users\USER\appdata\local\temp\arduino_build_947453\sketch\src\hal\hal_stm32\timers.h:122:26: note: suggested alternative: 'timer_t'
 extern void Step_Handler(stimer_t *htim);
                          ^~~~~~~~
                          timer_t
c:\users\USER\appdata\local\temp\arduino_build_947453\sketch\src\hal\hal_stm32\timers.h:122:36: error: 'htim' was not declared in this scope
 extern void Step_Handler(stimer_t *htim);
                                    ^~~~
c:\users\USER\appdata\local\temp\arduino_build_947453\sketch\src\hal\hal_stm32\timers.h:122:36: note: suggested alternative: 'tm'
 extern void Step_Handler(stimer_t *htim);
                                    ^~~~
                                    tm
c:\users\USER\appdata\local\temp\arduino_build_947453\sketch\src\hal\hal_stm32\timers.h:123:26: error: variable or field 'Temp_Handler' declared void
 extern void Temp_Handler(stimer_t *htim);
                          ^~~~~~~~
c:\users\USER\appdata\local\temp\arduino_build_947453\sketch\src\hal\hal_stm32\timers.h:123:26: error: 'stimer_t' was not declared in this scope
c:\users\USER\appdata\local\temp\arduino_build_947453\sketch\src\hal\hal_stm32\timers.h:123:26: note: suggested alternative: 'timer_t'
 extern void Temp_Handler(stimer_t *htim);
                          ^~~~~~~~
                          timer_t
c:\users\USER\appdata\local\temp\arduino_build_947453\sketch\src\hal\hal_stm32\timers.h:123:36: error: 'htim' was not declared in this scope
 extern void Temp_Handler(stimer_t *htim);
                                    ^~~~
c:\users\USER\appdata\local\temp\arduino_build_947453\sketch\src\hal\hal_stm32\timers.h:123:36: note: suggested alternative: 'tm'
 extern void Temp_Handler(stimer_t *htim);
                                    ^~~~
                                    tm
c:\users\USER\appdata\local\temp\arduino_build_947453\sketch\src\hal\hal_stm32\timers.h:131:9: error: 'stimer_t' does not name a type; did you mean 'timer_t'?
 typedef stimer_t stm32_timer_t;
         ^~~~~~~~
         timer_t
c:\users\USER\appdata\local\temp\arduino_build_947453\sketch\src\hal\hal_stm32\timers.h:137:8: error: 'stm32_timer_t' does not name a type; did you mean 'hal_timer_t'?
 extern stm32_timer_t TimerHandle[];
        ^~~~~~~~~~~~~
        hal_timer_t
In file included from C:\Users\USER\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\system/STM32F7xx/stm32f7xx_hal_conf_default.h:424,
                 from C:\Users\USER\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\system/STM32F7xx/stm32f7xx_hal_conf.h:13,
                 from C:\Users\USER\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\system/Drivers/STM32F7xx_HAL_Driver/Inc/stm32f7xx_hal.h:30,
                 from C:\Users\USER\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\system/Drivers/CMSIS/Device/ST/STM32F7xx/Include/stm32f7xx.h:219,
                 from C:\Users\USER\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\cores\arduino/stm32/stm32_def.h:36,
                 from C:\Users\USER\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\cores\arduino/stm32/clock.h:43,
                 from C:\Users\USER\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\cores\arduino/wiring_time.h:23,
                 from C:\Users\USER\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\cores\arduino/wiring.h:38,
                 from C:\Users\USER\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\cores\arduino/Arduino.h:32,
                 from c:\users\USER\appdata\local\temp\arduino_build_947453\sketch\src\hal\shared\marduino.h:36,
                 from c:\users\USER\appdata\local\temp\arduino_build_947453\sketch\src\hal\hal_stm32\hal.h:28,
                 from c:\users\USER\appdata\local\temp\arduino_build_947453\sketch\src\hal\hal.h:26,
                 from sketch\src\inc/MarlinConfig.h:30,
                 from sketch\src\Marlin.h:24,
                 from sketch\src\Marlin.cpp:31:
c:\users\USER\appdata\local\temp\arduino_build_947453\sketch\src\hal\hal_stm32\timers.h: In function 'uint32_t HAL_timer_get_count(uint8_t)':
[... 54 lines cut ...]
   return __HAL_TIM_GET_AUTORELOAD(&TimerHandle[timer_num].handle);
                                    ^~~~~~~~~~~
C:\Users\USER\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\system/Drivers/STM32F7xx_HAL_Driver/Inc/stm32f7xx_hal_tim.h:1351:6: note: in definition of macro '__HAL_TIM_GET_AUTORELOAD'
    ((__HANDLE__)->Instance->ARR)
      ^~~~~~~~~~
c:\users\USER\appdata\local\temp\arduino_build_947453\sketch\src\hal\hal_stm32\timers.h:159:36: note: suggested alternative: 'timeradd'
   return __HAL_TIM_GET_AUTORELOAD(&TimerHandle[timer_num].handle);
                                    ^~~~~~~~~~~
C:\Users\USER\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\system/Drivers/STM32F7xx_HAL_Driver/Inc/stm32f7xx_hal_tim.h:1351:6: note: in definition of macro '__HAL_TIM_GET_AUTORELOAD'
    ((__HANDLE__)->Instance->ARR)
      ^~~~~~~~~~
Multiple libraries were found for "TMCStepper.h"
 Used: F:\Arduino\libraries\TMCStepper
Multiple libraries were found for "SPI.h"
 Used: C:\Users\USER\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\libraries\SPI
Multiple libraries were found for "IWatchdog.h"
 Used: C:\Users\USER\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\libraries\IWatchdog
Multiple libraries were found for "U8glib.h"
 Used: F:\Arduino\libraries\arduino_645681
exit status 1
Error compiling for board 3D printer boards.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

shitcreek, 2019-10-03:
Looks like you are not using the correct library and you have multiple copies of libraries.
This video might help: https://www.youtube.com/watch?v=sB4lZ2zG5V8

**We're not a support group**
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2019-10-03 by shitcreek

SpecialMike, 2019-10-13:
I'd request that this issue is reopened. It seems like there is an issue compiling against the STM32 cores version 1.7.0. I had the same issue as OP, and downgrading to version 1.6.1 allows it to compile fine.

Spacemarine2018, 2019-10-24:
I have the same problem, seems to be a bug.

gattis, 2019-11-06:
same here, i tried watching the youtube link from the shitcreek auto shitresponse and it has nothing to do with the issue

shitcreek, 2019-11-07:
I'll reopen the issue if you are certain this is a bug and identical to this one.