MarlinFirmware/Marlin #15415: [BUG] SKR PRO + TMC5160 == TMC connection error

extesy, 2019-09-29:
### Description

In the latest version of Marlin (or TMCStepper) something got broken and now TMC5160s don't work with SKR PRO. I'm using Marlin's snapshot from 9/27/2019. I've also tried using TMCStepper library from https://github.com/bigtreetech/TMCStepper/archive/master.zip but that doesn't compile because TMCStepper's required version in Marlin is >=0.5 now.

### Steps to Reproduce

1. Use configs from my upload.
2. Place TMC5160s on SKR PRO 1.1, connect with REPRAP display.
3. Power the board.

**Expected behavior:** "Printer Ready" message.

**Actual behavior:** "TMC connection error" message.

#### Additional Information

Configs: [configs.zip](https://github.com/MarlinFirmware/Marlin/files/3666385/configs.zip)

ikn369330, 2019-09-29:
I have the same issue with the same setup, using latest marlin bugfix 2.0 TMC5160 v1,2 and SKR Pro v1.1 board
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/3666656/Config.zip)
SENDING:M122
		X	Y	Z	Z2	E
Enabled		false	false	false	false	false
Set current	800	800	800	800	1200
RMS current	792	792	792	792	1188
MAX current	1117	1117	1117	1117	1675
Run current	15/31	15/31	15/31	15/31	24/31
Hold current	7/31	7/31	7/31	7/31	12/31
Global scaler	133/256	133/256	133/256	133/256	128/256
CS actual	31/31	31/31	31/31	31/31	31/31
PWM scale	4294967295	4294967295	4294967295	4294967295	4294967295
stealthChop	true	true	true	true	true
msteps		0	0	0	0	0
tstep		4294967295	4294967295	4294967295	4294967295	4294967295
pwm
threshold
[mm/s]
OT prewarn	true	true	true	true	true
OT prewarn has
been triggered	false	false	false	false	false
off time	15	15	15	15	15
blank time	54	54	54	54	54
hysteresis
-end		12	12	12	12	12
-start		8	8	8	8	8
Stallguard thrs	0	0	0	0	0
DRVSTATUS	X	Y	Z	Z2	E
stallguard		*	*	*	*	*
sg_result	1023	1023	1023	1023	1023
fsactive		*	*	*	*	*
stst		*	*	*	*	*
olb		*	*	*	*	*
ola		*	*	*	*	*
s2gb		*	*	*	*	*
s2ga		*	*	*	*	*
otpw		*	*	*	*	*
ot		*	*	*	*	*
Driver registers:
		X	0xFF:FF:FF:FF	 Bad response!
		Y	0xFF:FF:FF:FF	 Bad response!
		Z	0xFF:FF:FF:FF	 Bad response!
		Z2	0xFF:FF:FF:FF	 Bad response!
		E	0xFF:FF:FF:FF	 Bad response!
Testing X connection... Error: All HIGH
Testing Y connection... Error: All HIGH
Testing Z connection... Error: All HIGH
Testing Z2 connection... Error: All HIGH
Testing E connection... Error: All HIGH

extesy, 2019-09-29:
@teemuatlut What do you think might be causing this?

emilhse, 2019-09-30:
I have exactly the same issue. i tried everything nothing helps. please help

alexsomesan, 2019-09-30:
Do you have `TMC_USE_SW_SPI` enabled in `Marlin/Configuration_adv.h` ?

extesy, 2019-09-30:
@alexsomesan Yes, I do. By the way, my full config files are attached.

ikn369330, 2019-09-30:
Yes I do and my configs are attached to my comment

Evg33, 2019-10-02:
Motor POW connected to PSU ?

extesy, 2019-10-03:
@Evg33 Yes.

ikn369330, 2019-10-03:
Yes

Evg33, 2019-10-03:
can you try 2 variants ?

1. delete caches in:
%USERPROFILE%\.platformio\.cache
%USERPROFILE%\.platformio\packages\
%USERPROFILE%\.platformio\platforms\
%USERPROFILE%\.platformio\lib\
Marlin\.pio\libdeps
and check again:
pio upgrade
pio update
pio run -t clean
pio run

2. with changes in platformio.ini in section [env:BIGTREE_SKR_PRO] at line 490.
-TMCStepper@>=0.5.0,<1.0.0
+TMCStepper=https://github.com/teemuatlut/TMCStepper/archive/master.zip
and check again

extesy, 2019-10-03:
@Evg33 Are you talking about https://github.com/teemuatlut/TMCStepper/commit/2d14f340a0966a16ac66ecfa435434aac3dd0f7d commit? Because that's the only meaningful commit that is not in the released version yet. Also, your step 1 is unnecessary because step 2 forces PlatformIO to redownload the library from a different source.

ikn369330, 2019-10-05:
I did as asked and surprising enough I still have the error

extesy, 2019-10-06:
@ikn369330 @emilhse Do you connect 12V power to the "POWER" input (screws 3 and 4)? Instead of that, try connecting it to "MOTOR POW" instead (screws 1 and 2). If you power the board from USB then you don't need external power for it, but you still need to power stepper drivers.

closed 2019-10-06 by extesy

ikn369330, 2019-10-06:
Why is this closed when the issue is not resolved, I e powered it as said and still have the error

extesy, 2019-10-06:
Because I opened this issue and it is resolved for me. Try also contacting BIGTREETECH support by email - they are usually very helpful.

dineshvyas, 2019-10-27:
I am a noob here. maybe jumpers can cause this issue. I checked my TMC5160 and it worked.

bywebberen, 2019-11-28:
I have the exact Same issue... I did remove the clk pin on the tmc5160 and was now wondering if I should solder in a diagnostic pin in the stepper driver to the second hole at the enable end. I saw a guy do this, but without the clk pin removed.
Setup
Skr v1.3 
Tmc5160 v1. 2

emilhse, 2019-11-29:
you have to power the board it will not work via usb supply. it will give you an error

bywebberen, 2019-11-29:
And do all kinds of things to the standard tmc5160 before it will with the
skr v1.3

Venlig hilsen / best regards
Niels Østbjerg
Syrenvænget 8
5792 Årslev

+45 40630201