MarlinFirmware/Marlin #15416: [BUG] EXTRUDERS 0 doesn't build anymore (no member named 'e_factor')

040medien, 2019-09-29:
### Description

A regression was introduced with commit 455dabb18334c89878a75be62bccebd9f537574d

Now, when you set EXTRUDERS to 0, you won't be able to build anymore. Relevant output:
```
Marlin/src/module/motion.cpp: In function 'void _internal_move_to_destination(const feedRate_t&)':
Marlin/src/module/motion.cpp:328:33: error: 'class Planner' has no member named 'e_factor'
   const float old_fac = planner.e_factor[active_extruder];

                                 ^
Marlin/src/module/motion.cpp:329:11: error: 'class Planner' has no member named 'e_factor'
   planner.e_factor[active_extruder] = 1.0f;
           ^
Marlin/src/module/motion.cpp:340:11: error: 'class Planner' has no member named 'e_factor'
   planner.e_factor[active_extruder] = old_fac;
           ^
Compiling .pio/build/megaatmega2560/src/src/module/stepper.cpp.o
*** [.pio/build/megaatmega2560/src/src/module/motion.cpp.o] Error 1
Compiling .pio/build/megaatmega2560/src/src/module/stepper/L6470.cpp.o
========================= [FAILED] Took 16.58 seconds =========================
```

### Steps to Reproduce

1. Get the latest bugfix-2.0.x from github.
2. Set this in Configuration.h:
`#define EXTRUDERS 0`
3. Build

**Expected behavior:** [What you expect to happen]
Build completes

**Actual behavior:** [What actually happens]
Build fails (see output above)

#### Additional Information

Not attaching Configurations, since I only changed one line vs current HEAD (43d6e9f).

040medien, 2019-09-29:
This is fixed. There are probably more bugs like this elsewhere (e.g. UBL), but I'm not using these features on my CNC.

closed 2019-09-29 by 040medien

thinkyhead, 2019-09-29:
Being able to build Marlin with `EXTRUDERS` set to 0 has been a longstanding goal we thought we would never reach, so even just being at the point we are is gratifying.

040medien, 2019-09-30:
Thank you for making this happen! The MPCNC community is very excited:
https://www.v1engineering.com/forum/topic/omg-extruders-0/