MarlinFirmware/Marlin #15419: [BUG] Build error M48.cpp

hapklaar, 2019-09-29:
`
Marlin\src\gcode\calibrate\M48.cpp: In static member function 'static void GcodeSuite::M48()':
Marlin\src\gcode\calibrate\M48.cpp:179:24: error: ambiguous overload for 'operator*=' (operand types are 'xy_float_t' {aka 'XYval<float>'} and 'double')
               next_pos *= 0.8;
               ~~~~~~~~~^~~~~~
In file included from Marlin\src\gcode\calibrate\../../inc/MarlinConfig.h:42,
                 from Marlin\src\gcode\calibrate\M48.cpp:23:
Marlin\src\gcode\calibrate\../../inc/../core/types.h:260:16: note: candidate: 'XYval<T>& XYval<T>::operator*=(const float&) [with T = float]'
   FI XYval<T>& operator*=(const float &v)               { x *= v;    y *= v;    return *this; }
                ^~~~~~~~
Marlin\src\gcode\calibrate\../../inc/../core/types.h:261:16: note: candidate: 'XYval<T>& XYval<T>::operator*=(const int&) [with T = float]'
   FI XYval<T>& operator*=(const int &v)                 { x *= v;    y *= v;    return *this; }
                ^~~~~~~~
*** [.pio\build\LPC1768\src\src\gcode\calibrate\M48.cpp.o] Error 1
`

arminth, 2019-09-29:
Can confirm that. Had to disable M48 command.

thinkyhead, 2019-09-29:
Patched with f307ff68f7. Thanks for the report!

closed 2019-09-29 by thinkyhead