MarlinFirmware/Marlin #15421: SKR Pro V1.1 build error when enabling "FLASH_EEPROM_EMULATION"

xeddog, 2019-09-29:
### Description

After latest pull (09-28), build fails when "#define FLASH_EEPROM_EMULATION  // Use Flash-based EEPROM emulation" is enabled.  Build runs ok when it is disabled.

### Steps to Reproduce

1. enable the option in pins_BIGTREE_SKR_PRO_V1.1.h
2.  run build

**Expected behavior:** [What you expect to happen]
Successful Build

**Actual behavior:** [What actually happens]
Build fails-  See attached zip for details

#### Additional Information

Attached zip file contains configuration.h, configuration.adv.h, pins_BIGTREE_SKR_PRO_V1.1.h, and a screenshot showing the detailed errors.

brianredbeard, 2019-09-29:
Possibly related to #15371

xeddog, 2019-09-29:
Possible, but it looks like they are both for different boards with different processors.  I will have to experiment with the flash to SD option again.  I thought I had it working, but then it just stopped so maybe it is the same problem as the other boards.  But this build error is different.

brianredbeard, 2019-09-29:
@xeddog there was a bunch of work done yesterday across some of the SKR / Bigtreetech platforms.

If you look at the commits added in https://github.com/MarlinFirmware/Marlin/pull/15398 you'll see a bunch of changes overall to the STM32F103 based platforms.

Evg33, 2019-10-02:
BIGTREE_SKR_PRO          SUCCESS
```
#define EEPROM_SETTINGS
#define EEPROM_CHITCHAT
#define FLASH_EEPROM_EMULATION

```

xeddog, 2019-10-03:
I have run a couple of pulls and now it looks like TWA (Trouble Went Away).  Resolved as far as I'm concerned.

closed 2019-10-03 by xeddog