MarlinFirmware/Marlin #15422: [QUESTION] Responses in Serial communication

MarkMan0, 2019-09-29:
I'm in the process of writing a program, which will communicate with Marlin from my PC, monitor the status, checksums etc.

I would like to know, what kind of messages can I expect from the printer. What I've got for now are:
1. ok - indicates that I can send the next command
2. echo:....  - regular message, nothing to do with it
3. temp reports - M155, ...
4. position reports - G28, ....
5. Errors
6. Resend requests - when checksum fails

Is there anything else? Not talking about user commands wih **M118**. 
Can Marlin send, for example: _//message_ by itself?

If this is not yet documented anywhere, I would gladly do it...

On a side note, I'm using and Ender 3 motherboard, and it sometimes send some characters twice, like: 
_T:-15.00 /0.00 B::-15.00 /0.00 @:0 B@:0_ ---- Two colons
_T:-15..00 /0.000 B:-15.00 /0.00 @:0 B@:0_    ---- Two decimal points

Is this a known bug, or should I lower my baudrate(115200)?

thinkyhead, 2019-09-29:
For a master course on how to parse Marlin messages you should look at the source code of OctoPrint. It handles those that come with `echo:` or `//action:` on the front, parses the output of all the native commands, plus plugins are able to parse the output of `M420 V` and more.

boelle, 2019-10-24:
please ask here instead

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

this list is for bugs only (and FR's)

closed 2019-10-24 by boelle