MarlinFirmware/Marlin #15426: [BUG] Compiling with G26_MESH_VALIDATION but not ARC_SUPPORT fails

brianredbeard, 2019-09-30:
### Description

Seemingly, after pulling in 50e4545255605eb506c20eb107270038b0fe7bdb compiling with `G26_MESH_VALIDATION` but failing to enable `ARC_SUPPORT` fails.  Re-enabling `ARC_SUPPORT` causes the build to succeed.

### Steps to Reproduce

1.Perform a clean on the project
2. Enable `G26_MESH_VALIDATION`
3. Disable `ARC_SUPPORT`
4. Compile

Capture the following error:

```
Marlin/src/gcode/bedlevel/G26.cpp: In static member function 'static void GcodeSuite::G26()':
Marlin/src/gcode/bedlevel/G26.cpp:803:97: error: too many initializers for 'xy_float_t {aka XYval<float>}'
           xy_float_t p = { circle.x + _COS(ind    ), circle.y + _SIN(ind    ), g26_layer_height },
                                                                                                 ^
Marlin/src/gcode/bedlevel/G26.cpp:804:97: error: too many initializers for 'xy_float_t {aka XYval<float>}'
                      q = { circle.x + _COS(ind + 1), circle.y + _SIN(ind + 1), g26_layer_height };
```

#### Additional Information

Attached is the file `build.zip` which contains the configurations and a logs of successful and failed builds.
[build.zip](https://github.com/MarlinFirmware/Marlin/files/3668726/build.zip)

DrDitto, 2019-09-30:
I have  no arc support and G26_MESH_VALIDATION enabled without compile error

oliver-eifler, 2019-09-30:
if've got the same errors -never tried it with ARC_SUPPORT_ENABLED...
changed (line 803/804 in gcode/G26.cpp)
```          
      xy_float_t p = { circle.x + _COS(ind    ), circle.y + _SIN(ind    ), g26_layer_height },
                     q = { circle.x + _COS(ind + 1), circle.y + _SIN(ind + 1), g26_layer_height };
```
to      

```         
     xyz_float_t p = { circle.x + _COS(ind    ), circle.y + _SIN(ind    ), g26_layer_height },
                      q = { circle.x + _COS(ind + 1), circle.y + _SIN(ind + 1), g26_layer_height };
```
and it compiled fine, anybody could confirm this?

closed 2019-10-02 by thinkyhead