MarlinFirmware/Marlin #15427: Z_Stepper_Auto_Align

jibbersmcgee, 2019-09-30:
Trying to compile Marlin 2.0 for an SKR 1.3 with 5x TMC2130 Stepper drivers. I want to make use of the dual Z endstop leveling. No matter what I try, I encounter the below. Hopefully it's something simple and not a bug but would appreciate a sanity check.

**Expected behavior:**Compilation

**Actual behavior:**Compliation terminates with the following errors

In file included from Marlin\src\gcode\calibrate\../../inc/MarlinConfigPre.h:44,
                 from Marlin\src\gcode\calibrate\../../inc/MarlinConfig.h:28,
                 from Marlin\src\gcode\calibrate\G34_M422.cpp:3:
c:\users\small\desktop\legit\marlin\configuration_adv.h:605:44: error: expected unqualified-id before '{' token
   #define Z_STEPPER_ALIGN_XY {  10, 290 }, { 150,  10 }, { 290, 290 }  // Set number of iterations to align
                                            ^
Marlin\src\gcode\calibrate\G34_M422.cpp:28:43: note: in expansion of macro 'Z_STEPPER_ALIGN_XY'
 constexpr xy_pos_t sanity_arr_z_align[] = Z_STEPPER_ALIGN_XY;
                                           ^~~~~~~~~~~~~~~~~~
c:\users\small\desktop\legit\marlin\configuration_adv.h:605:56: error: expected unqualified-id before ',' token
   #define Z_STEPPER_ALIGN_XY {  10, 290 }, { 150,  10 }, { 290, 290 }  // Set number of iterations to align
                                                        ^
Marlin\src\gcode\calibrate\G34_M422.cpp:28:43: note: in expansion of macro 'Z_STEPPER_ALIGN_XY'
 constexpr xy_pos_t sanity_arr_z_align[] = Z_STEPPER_ALIGN_XY;
                                           ^~~~~~~~~~~~~~~~~~
c:\users\small\desktop\legit\marlin\configuration_adv.h:605:58: error: expected unqualified-id before '{' token
   #define Z_STEPPER_ALIGN_XY {  10, 290 }, { 150,  10 }, { 290, 290 }  // Set number of iterations to align
                                                          ^
Marlin\src\gcode\calibrate\G34_M422.cpp:28:43: note: in expansion of macro 'Z_STEPPER_ALIGN_XY'
 constexpr xy_pos_t sanity_arr_z_align[] = Z_STEPPER_ALIGN_XY;
                                           ^~~~~~~~~~~~~~~~~~
Marlin\src\gcode\calibrate\G34_M422.cpp:29:41: error: static assertion failed: Z_STEPPER_ALIGN_XY requires two {X,Y} entries (Z and Z2).
 static_assert(COUNT(sanity_arr_z_align) == Z_STEPPER_COUNT,
compilation terminated due to -fmax-errors=5.
*** [.pio\build\LPC1768\src\src\gcode\calibrate\G34_M422.cpp.o] Error 1

boelle, 2019-10-01:
please also attach the 2 config files just as the issue template told you too (configuration.h and configuration_adv.h)

Aurawin, 2019-10-05:
In Configuration_adv.h

Here are my settings.  There are two arrays.  It's for the left than the right stepper.
left stepper X = 40, Y=175
right stepper X=310,  Y=175

#define Z_STEPPER_AUTO_ALIGN
  #define Z_STEPPER_ALIGN_X { 40, 310 }
  #define Z_STEPPER_ALIGN_Y { 175, 175 }
  #define Z_STEPPER_ALIGN_ITERATIONS 3
  #define G34_MAX_GRADE  2

stefi01, 2019-10-08:
im getting error Z_STEPPER_ALIGN_XY requires two {X,Y} entries (Z and Z2).
i know its related to this but no idea how to set it, everything compliles until i try add the align feature.

 static_assert(COUNT(sanity_arr_z_align) == Z_STEPPER_COUNT,

 ^~~~~~~~~~~~~

In file included from c:\users\skipp\appdata\local\temp\arduino_build_120925\sketch\src\inc\MarlinConfigPre.h:44:0,

                 from c:\users\skipp\appdata\local\temp\arduino_build_120925\sketch\src\inc\marlinconfig.h:28,

                 from sketch\src\gcode\calibrate\G34_M422.cpp:23:

c:\users\skipp\appdata\local\temp\arduino_build_120925\sketch\configuration_adv.h:605:73: error: too many initializers for 'xy_pos_t [2] {aka XYval<float> [2]}'

   #define Z_STEPPER_ALIGN_XY { {  10, 290 }, { 150,  10 }, { 290, 290 } }  // Set number of iterations to align

                                                                         ^

sketch\src\gcode\calibrate\G34_M422.cpp:58:46: note: in expansion of macro 'Z_STEPPER_ALIGN_XY'

 xy_pos_t z_auto_align_pos[Z_STEPPER_COUNT] = Z_STEPPER_ALIGN_XY;

[marlin_2.0_rambo.zip](https://github.com/MarlinFirmware/Marlin/files/3702755/marlin_2.0_rambo.zip)

Patag, 2019-10-08:
You need to keep 2 pairs of coordinates only for a dual Z config
`#define Z_STEPPER_ALIGN_XY { { 10, 290 }, { 150, 10 } }` for example

boelle, 2019-10-12:
@jibbersmcgee still having issues?

boelle, 2019-10-12:
@jibbersmcgee 

look at the fail:

c:\users\small\desktop\legit\marlin\configuration_adv.h:605:44: error: expected unqualified-id before '{' token

boelle, 2019-10-22:
as there  has been no acticity and a fix has been provided i will close this one

we can always reopen

closed 2019-10-22 by boelle