MarlinFirmware/Marlin #15429: [BUG] (Compilation error when including Servo Z Probe - Board SKRV1.3 )

himanshu0192, 2019-09-30:
Hi,
I am getting a compilation error while building the latest marlin firmware dated 28/08/2019. 
The issue comes when I enable 
#define Z_PROBE_SERVO_NR 0       // Defaults to SERVO 0 connector.
#define Z_SERVO_ANGLES { 13, 100 } // Z Servo Deploy and Stow angles
If I disable the above lines then the build is successful.
The build log is as follows
[image]

Project files
[Uploading Marlin-bugfix-2.0.x.zip…]

Config Files
[Config_files.zip](https://github.com/MarlinFirmware/Marlin/files/3667871/Config_files.zip)
 
Using Atom + Platform.io on Windows 10

tvixen, 2019-10-01:
Please use the default template for submitting errors/bugs. 
It's difficult for anyone to guess your hardware setup, even with the config files. 
Do you your have a bedleveling device of some kind ?

sjasonsmith, 2019-10-05:
Your config builds just fine for me. I built on top of the latest bugfix-2.0.x commit, 5d81d7cefe98e63442e3babee6b21b2df8f02fa8.

boelle, 2019-10-12:
as this builds just fine i will close

we can always reopen

closed 2019-10-12 by boelle