MarlinFirmware/Marlin #15430: [FR] Add laser/tool offset from nozzle

LinoBarreca, 2019-09-30:
### Description

Actually you can specify an X/Y/Z offset for the zprobe (with the nozzle at 0,0).
If you have a CNC/Laser attached to your printer head there isn't a way to "automatically check" if it's use is in the bounds (of the printing/engraving/machining area) or not.

Actually laser/tool isn't managed as a nozzle and this is problematic because I can tell to marlin to
extrude L mm from nozzle A at position X,Y
but I can't tell marlin to
engrave W power with laser L at position X,Y

position of laser actually isn't relative to the laser itself but absolute.

### Steps to Reproduce

Give a way for the user to define the laser as an extruder (along with its offsets).

#### Additional Information

none

AnHardt, 2019-09-30:
**Length** can't be compared to **Watt**. It must be compared to __Watt\*second__.

To make pile of filament on a spot you need to extrude some mm³ of filament (filament cross section \* filament length).
To reach a determined *blackness* you have to apply a amount of energy per surface [Ws / dotsrea]. The laser has to apply a pulse of given strength in Watt lasting the duration time. (Or N pulses each having a energy of Watt\*seconds).
Then extruder_steppers *steps/unit* would then be lasers *laser_on_time/unit* or *pulses/unit*.

Likely a PWM for the laser reacts much to slow. *Puls Density Modulation* or *DeltaSigmaModulation*. could be a better choice.

LinoBarreca, 2019-09-30:
> **Length** can't be compared to **Watt**. It must be compared to **Watt*second**.

you didn't get the point of what I meant.

I meant to keep the "parameters" of the G0-G3 commands but meant in a different way when the "extruder" is a laser (power 0 to 255) or a tool (RPM 0 to 255) .
Actually the laser power is regulated through PWM (0 to 255), but... instead of managing this as "laser" if it was possible to manage it like an extruder (same command "logic") it would be better since you didn't have to make the calculations about offsets and use "normal" extrude commands.

but yeah I agree with you about the watts/seconds or even better watts/m since parameter E of the G0-G3 is actually a LENGTH (to extrude) between start and end...But to make it simple if that "length" would be "laser power" or "spindle RPM" between the start and end point you could say "go to here, with this power, then to here with this power" and so on... without changing the G0-G3 commands at all...

I could "slice" a 1 layer stl and have something etched out of the box.

LinoBarreca, 2019-09-30:
...and no. the pwm doesn't react too slow. [I already use it](https://cdn.thingiverse.com/assets/b7/32/c7/01/0a/featured_preview_20190929_114200.jpg)...I just have to mess with the gcode to have a decent result.

AnHardt, 2019-09-30:
> ... better watts/m ...

Again wrong. Ws/m!

LinoBarreca, 2019-09-30:
> 
> 
> > ... better watts/m ...
> 
> Again wrong. Ws/m!

Again you focus on unit measures than concept. you don't have to consider watts at all but rather "PWM value" you don't really want to define all the parameters of the laser (absolute power, wavelength, radius and so on) you just need "how much should I set the PWM when moving to there?" then if you connect a 10w laser or a CO2 100W lasers is user's business, not related to marlin.

AnHardt, 2019-09-30:
Because Marlin moves at  different speeds in one move, it accelerates at the beginning and brakes at the end, a constant power over the complete move does not work. You get "blacker" lines at the start and end of a that lines where the laser moves slower.

But it's your choice. Do you wand a simple or a working solution. 
In my opinion we already had enough of the simple ones.

LinoBarreca, 2019-09-30:
acceleration can be reflected into the PWM anyway. if you set your PWM to 100 and printing speed to 70mm/s you know that your pwm must be 100 "at linear speed". if you, obviously, have acceleration (and inertia) your "fed" PWM will be lower at the start/end accordingly (in the same way you extrude faster as your print head go faster)

LinoBarreca, 2019-09-30:
> But it's your choice. Do you wand a simple or a working solution.
> In my opinion we already had enough of the simple ones.

You don't need over-complicate solutions for simple things.
You can define the E parameter as you want as soon as changing the E will change accordingly the laser power in the same way they extrude more material.
call it mm/m gr/cm molecules/mm photons/s photons/m I don't care....but having to define the laser parameters in marlin looks like an overkill to me (not to consider that the more parameters you have the more complicate the math is....and, in the end, marlin will have to feed a PWM value to the power pin so why making it unnecessarily complicate?)

LinoBarreca, 2019-09-30:
(and anyway, it's the same parameter "logic" to the M3 command)

shitcreek, 2019-10-08:
We've got a PR for reworking the laser/spindle stuff over here #15335 - have a look.
We should definitely have an offset option for the laser/spindle.

thinkyhead, 2019-10-08:
It is not intended for Marlin to have a Laser and a Nozzle on the same device.

LinoBarreca, 2019-10-09:
> It is not intended for Marlin to have a Laser and a Nozzle on the same device.
as once it was not intended to have more than one nozzle on the same device...then support came and multi-color printers started to roll out.
Add proper support to the firmware and manufacturers will add a 10$ laser head to increase the price of 50$...
In a matter of time slicers will add laser ironing/postprocessing to the piece.

AnHardt, 2019-10-09:
All "tools" (nozzles, lasers, cutters, probes, ...) should have a "offset" from a "carriage" reference point (something as invariant as possible). When working with tool-changers this becomes essential.
Offset from "nozzle" makes low sense if **the** "nozzle" is also a variable.

LinoBarreca, 2019-10-09:
> All "tools" (nozzles, lasers, cutters, probes, ...) should have a "offset" from a "carriage" reference point (something as invariant as possible). When working with tool-changers this becomes essential.
> Offset from "nozzle" makes low sense if **the** "nozzle" is also a variable.

Man...Are you here to troll or what?
All your comments are like "hey, you say bullshit, I own the truth".

On Marlin the nozzle is, by definition, at (0,0,0) all other measurements (probe and other nozzles) take the first nozzle as reference....and anyway, the first nozzle is fixed on the carriage, so having that as a reference point, actually, make  sense. Stop commenting if you don't have something useful besides "I know better". Yes, with tool changers and stuff you would need a reference to the carriage but actually nozzle of E0 is at (0,0,0)

AnHardt, 2019-10-09:
Probably you are right.
I'm currently not as deep in the code as i used to be. I can't keep up. That's frustrating to me.
It's caused by my health and too much time consuming crappy contributions to this issue tracer.
But my [experience](https://github.com/MarlinFirmware/Marlin/graphs/contributors) with Marlin and a bit more distance to the actual code enables me to see some of the conceptual problems with Marlin.

"State of the art" implies "What's good enough today will not be sufficient in the future."

If you can't find any value in my contributions feel free to [block](https://help.github.com/en/articles/blocking-a-user-from-your-personal-account) me.

LinoBarreca, 2019-10-09:
> 
> "State of the art" implies "What's good enough today will not be sufficient in the future."

Probably you are not into "real life development" and, most in general, into "one step at a time".

"Real life" implies: "When the development goal involves too much work and you don't have enough resources you are unlikely to even start it."
Going from "no usable implementation" (actually) to "state of art" (what you suggest and everyone aims to) just doesn't work and usually involves changes so big that get ignored.

Do you want to change 90% of the code from the base (basically rewriting it) to properly implement tool change, with multiple tools and implement each tools correctly? Good for you.
As far as I see you are just making the professor telling how things should be done if they were perfect and "feature" complete (as we didn't already know). I highly doubt someone would waste its time to write (and submit) a patch which does that.

In the "real world", most likely, if someone wastes all that time to implement and rewrite things as they should be, he does a fork and starts another "project" because the impacts would be so big that even merging it would require months and break another dozen of features here and there.
Now...you are really amazing, thanks for telling us what the state of art would be.
Now let's get back to the "what steps can be done" to get a bit close to the state of art without having to rewrite the whole code at once.

@shitcreek 's proposed patch looks very good and promising. It will NOT make the software the state of the art but it brings "meh" to "okay, it's better" and one step closer how it should be.

closed 2020-06-30 by github-actions[bot]