MarlinFirmware/Marlin #15432: [BUG] Probing before print failed, display says "printing"

AndKe, 2019-09-30:
### Description

if bed probing fail, printer's display is still displaying "printing..."
Heaters remain on.

### Steps to Reproduce
let probing fail, while probing is in the start of the print...

**Expected behavior:** [What you expect to happen]

display should say "probing failed"  , heaters be switched off
**Actual behavior:** [What actually happens]

#### Additional Information
Octoprint terminal: 

Recv: echo:busy: processing
Recv:  T:177.90 /200.00 B:74.46 /75.00 @:127 B@:127
Recv: echo:busy: processing
Recv:  T:183.87 /200.00 B:74.69 /75.00 @:127 B@:127
Recv: echo:busy: processing
Recv: Error:Probing Failed
Changing monitoring state from "Printing" to "Error: Probing Failed"
Changing monitoring state from "Error: Probing Failed" to "Offline (Error: Probing Failed)"
Connection closed, closing down monitor
Connecting to: /dev/ttyUSB0

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3670504/Configuration.zip)
[gcode.zip](https://github.com/MarlinFirmware/Marlin/files/3670517/gcode.zip)

reloxx13, 2019-09-30:
link cause this also happens if not failing

#13534

boelle, 2019-10-12:
@AndKe is this still an issue?

AndKe, 2019-10-12:
Yes.

boelle, 2019-10-24:
i have also found out that if i cancel a print in octoprint the progress bar stays where it is
so something is wrong with either octoprint or marlin, will dive down in octoprint and see what happens

but yeah if you print from sd card and start with a probe that fails it might be that is continues with the print? or does it just say  print?

@shitcreek does any of this make sense?

boelle, 2019-11-03:
@AndKe many updates have been added in the last 22 days, is the issue still there in 2.0.x ?

AndKe, 2019-11-03:
@boelle I can't say for sure. Need to diff my config files and rebase everything on current master to know.

boelle, 2019-11-11:
@AndKe did you have time to diff and rebase?

AndKe, 2019-11-21:
@boelle 
I tried to reproduce today, with bilinear 9-point levelling, I did not manage to make it fail... 
I let the BLtouch probe touch something >5mm too early at random points , and yet, the levelling was apparently accepted.
so the test is inconclusive.

boelle, 2019-11-25:
@AndKe you think the issue will ever surface again?

AndKe, 2019-11-25:
hard to tell, given that levelling did not fail even if I tried to make it, but this issue may be closed, and reopened if it proves to be an issue still...

boelle, 2019-11-25:
sounds like a plan

if a reopen is needed just put @ in front of my username as i might get noted for sure

closed 2019-11-25 by boelle